ginkage/MHI-AC-Ctrl-ESPHome #1: p

DYLaKo, 2021-05-25:


ginkage, 2021-05-25:
Should be possible, the values themselves are there, see [here](https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/blob/981729c234bf870b70cdb76b14b3a3b90fd70528/mhi_ac_ctrl.h#L236) for example. These values are still received, they're just not exposed, and it should be a fairly trivial modification to add them as additional sensors.
Unfortunately I'm too far from my home where my aircons are, and unlikely to visit it in a few months, so I can't test modifications that would expose these. But the source code is open, so feel free to contribute or bring the community's attention to the project, I'm sure somebody will happily take on this task.

DYLaKo, 2021-05-26:
I should be able to do some modifications myself if I know where to start. :)
Is it possible to help me out with some code/modification for 1 sensor, so I can do the rest myself? I know you cannot test, but I can test for you. :)

ginkage, 2021-05-26:
Sounds good. :)
I've created a new [branch](https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/tree/sensors), and pushed two sample [commits](https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/commit/17541ed8087dc69d7c380a61d29eaae5d285eb5f) [there](https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/commit/4525757c7abd5e9985d0ea9177fd135fa71f5dbe), to demo adding an integer sensor, a float sensor, and a binary sensor, with icons and measurement units.
Not sure if this is going to be enough, but it's worth a try.

ginkage, 2021-05-26:
Note that it might be a better idea to expose these as attributes rather than sensors. I'll look into it a bit later.

DYLaKo, 2021-05-26:
I will try your new branch this weekend and post the results here.

ginkage, 2021-06-06:
On a related note, attributes seem to be impossible with ESPHome: https://github.com/esphome/feature-requests/issues/115

deku-m, 2021-06-18:
@ginkage 
Did you also notice the new changelog 1.19.0
https://esphome.io/changelog/v1.19.0.html (attributes for home assistant sensor?

ginkage, 2021-06-18:
@deku-m Yes, I saw that one, I believe that's the exact opposite: it's for reading attributes from existing entities, rather than exposing my own attributes.

deku-m, 2021-06-18:
Ah damn to bad haha i just had to upgrade and saw the changelog and remember your feedback xD

DYLaKo, 2021-06-18:
Still haven't found time to test.. sorry
Will test next week to see the results.

ginkage, 2021-07-03:
Just in case, merged the main branch in, so that you can try it with ESPHome 1.19.4 for whenever you have time.

OperKH, 2021-07-11:
> I've created a new [branch](https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/tree/sensors), and pushed two sample [...]

Made a tiny build fix in #9 PR and its works!

[image]

ginkage, 2021-07-11:
I believe the PR mentioned above is not required though.

SomeSander, 2021-11-11:
Hi @ginkage ,

The new branch works great!! I also saw other people in the Home Assistant Dutch Facebook group confirming this. Would it be helpful to add it to the master for future users with a switch to enable/disable technical logging? In my case I did make some adjustments, as the ID and name in home assistant isn't really clear, as it belongs to the MHI device. Also I have two MHI airco's hence adjusted the names. 

Do you know if the ID can be set also directly in the custom platform? I couldn't find it at: https://esphome.io/components/sensor/custom.html Now it is derived from the name, but I think it is helpful to keep them separate. Next to that it would help to add a variable with the the Airco name, so the climate and sensors are based on the same prefix.

`climate:
  - platform: custom
    lambda: |-
      auto mhi_ac_ctrl = new MhiAcCtrl();
      App.register_component(mhi_ac_ctrl);
      return {mhi_ac_ctrl};

    climates:
      - name: "MHI-AC-Ctrl Downstairs"
        id: mhi_ac_ctrl_downstairs

sensor:
  - platform: custom
    lambda: |-
      return ((MhiAcCtrl*)mhi_ac_ctrl_upstairs)->get_sensors();

    sensors:
      - name: "MHI-AC-Ctrl Downstairs Error code"
      - name: "MHI-AC-Ctrl Downstairs Outdoor temperature"

binary_sensor:
  - platform: custom
    lambda: |-
      return ((MhiAcCtrl*)mhi_ac_ctrl_upstairs)->get_binary_sensors();

    binary_sensors:
      - name: "MHI-AC-Ctrl Downstairs Defrost"`

DYLaKo, 2021-11-11:
> edit: While the sensors are only a subset (only noticed the remarks late :) ) it works great. Additional sensors should be trivial, but looking at the .h file it is complex for me as I don't have experience here. Wondering if @DYLaKo already already have included all sensors.

To be honest, I had some turbulent months in which I used my spare time to relax instead of working on my home automation, so both MHi's in my house are still running with the code from the original MHI-AC-Ctrl project.

Hope to use this is in the near future.

ginkage, 2021-11-30:
> Would it be helpful to add it to the master for future users with a switch to enable/disable technical logging?

I'll get to that, hopefully soon enough.

SomeSander, 2021-11-30:
> > Would it be helpful to add it to the master for future users with a switch to enable/disable technical logging?
> 
> I'll get to that, hopefully soon enough.

No issues. Great that your are willing already, thanks!

synegic, 2022-04-18:
I've added some more sensors in my pull request; https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/pull/16. If you want, you could test it out and report back if it works for you. Also, let me know if you are still missing any sensors, I'll happily try to add them.

closed 2022-04-23 by ginkage