ginkage/MHI-AC-Ctrl-ESPHome #2: A few questions

aidbish, 2021-06-08:
hi, thanks for converting to esphome, much preferred

wondering if you could answer the following

Do i need to add the api: part in the yaml to show up in HA

ginkage, 2021-06-08:
It seems to be needed to show up in the dashboard, and to view logs in the ESPHome component of HA, but I believe it's not needed to just integrate the AC entities in HA.
Not sure about the state change (unfortunately I'm far away from my setup to test it), there may be some minor bug...

ginkage, 2021-07-03:
Anyway, "api:" is now included in the default configuration.
Can't check up on the "cool" by default, feel free to raise a separate issue for that if this is still the case after the upgrade to 1.19.4 (can't check at the moment, as I'm away from my hardware).

closed 2021-07-03 by ginkage

xz0609, 2021-07-08:
> hi, thanks for converting to esphome, much preferred
> 
> wondering if you could answer the following
> 
> Do i need to add the api: part in the yaml to show up in HA
> 
> Upon restarting the esp8266, the state changes to cool in HA even though not on.?

esphome:
  name: ${device_name}
  platform: ESP8266
  board: d1_mini
  includes:
    - mhi_ac_ctrl/mhi_ac_ctrl.h
    - mhi_ac_ctrl/MHI-AC-Ctrl-core.h
    - mhi_ac_ctrl/MHI-AC-Ctrl-core.cpp
  on_boot:
    priority: 250
    then:
      - climate.control: 
          id: mhi_ac_ctrl
          mode: 'OFF'
          target_temperature: 24°C
          fan_mode: 'AUTO'

ginkage, 2021-07-08:
This might be a workaround, though ideally it shouldn't be required.
Do you mind checking whether removing the "restore" logic in mhi_ac_ctrl.h (in setup() method) would achieve a similar result?

xz0609, 2021-07-09:
> This might be a workaround, though ideally it shouldn't be required.
> Do you mind checking whether removing the "restore" logic in mhi_ac_ctrl.h (in setup() method) would achieve a similar result?

I tried to comment 'restore' code in  setup() function. The content is as follows:
```
public:
    void setup() override
    {
        this->current_temperature = NAN;
        // restore set points
        // auto restore = this->restore_state_();
        // if (restore.has_value()) {
            // restore->apply(this);
        // } else {
            // restore from defaults
            // this->mode = climate::CLIMATE_MODE_OFF;
            // initialize target temperature to some value so that it's not NAN
            // this->target_temperature = roundf(clamp(
                // this->current_temperature, this->minimum_temperature_, this->maximum_temperature_));
            // this->fan_mode = climate::CLIMATE_FAN_AUTO;
            // this->swing_mode = climate::CLIMATE_SWING_OFF;
        // }
        // Never send nan to HA
        if (isnan(this->target_temperature))
            this->target_temperature = 24;

        mhi_ac_ctrl_core.MHIAcCtrlStatus(this);
        mhi_ac_ctrl_core.init();
    }
```

It can't change from heat_cool to off.
after OTA firmware, The AC state is still heat_cool in HA