ginkage/MHI-AC-Ctrl-ESPHome #3: Does not compile in ESPhome

ACMHI, 2021-06-27:
Hi, thanks for transferring this to ESPhome. Unfortunately it does not compile in my ESPhome V 1.19.4. I tried with ESP8266 Wemos D1 mini. The Error report:

In file included from src/main.cpp:17:0:
src/mhi_ac_ctrl.h: In member function 'virtual void MhiAcCtrl::control(const esphome::climate::ClimateCall&)':
src/mhi_ac_ctrl.h:323:35: error: switch quantity not an integer
             switch (this->fan_mode) {

Do you have an idea whats the problem?

Thank you very much!
Johannes

ginkage, 2021-06-27:
Hi, @ACMHI thank you for reporting!
I've just pushed a commit that might fix it (I'm far away from my A/C, so can't test atm), please check!
Looks like in the latest ESPHome they've made "fan_mode" an "optional", so it was missing a single asterisk.

ACMHI, 2021-06-28:
Hi ginkage,

cool, thanks, now it compiles without issues :-)

ginkage, 2021-06-28:
Confirmed to be fixed.

closed 2021-06-28 by ginkage