ginkage/MHI-AC-Ctrl-ESPHome #4: [ERROR] during code compile in ESPHome

lifeandfate1, 2021-06-28:
Hi

I have been attempting to setup 3 devices using your ESPHome integration. 
Unfortunately when I try to flash it and upon first compile of the app I get the following errors relating to method depreciated.
I'm just wondering if there is anything I can do to remove these errors as shows below .

INFO Reading configuration /config/esphome/lr_mhi_ac_ctrl.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/lr_mhi_ac_ctrl
Processing lr_mhi_ac_ctrl (board: d1_mini; framework: arduino; platform: platformio/espressif8266@2.6.2)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20704.0 (2.7.4) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Library Manager: Installing Update
Library Manager: Already installed, built-in library
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
Compiling /data/lr_mhi_ac_ctrl/.pioenvs/lr_mhi_ac_ctrl/src/main.cpp.o
Generating LD script /data/lr_mhi_ac_ctrl/.pioenvs/lr_mhi_ac_ctrl/ld/local.eagle.app.v6.common.ld
Compiling /data/lr_mhi_ac_ctrl/.pioenvs/lr_mhi_ac_ctrl/lib4d9/ESP8266WiFi/BearSSLHelpers.cpp.o
In file included from src/main.cpp:10:0:
src/mhi_ac_ctrl.h: In member function 'virtual void MhiAcCtrl::control(const esphome::climate::ClimateCall&)':
src/mhi_ac_ctrl.h:307:35: error: switch quantity not an integer
             switch (this->fan_mode) {
                                   ^
src/mhi_ac_ctrl.h: In member function 'virtual esphome::climate::ClimateTraits MhiAcCtrl::traits()':
src/mhi_ac_ctrl.h:352:43: warning: 'void esphome::climate::ClimateTraits::set_supports_auto_mode(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:54): This method is deprecated, use set_supported_modes() instead [-Wdeprecated-declarations]
         traits.set_supports_auto_mode(true);
                                           ^
src/mhi_ac_ctrl.h:353:43: warning: 'void esphome::climate::ClimateTraits::set_supports_cool_mode(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:56): This method is deprecated, use set_supported_modes() instead [-Wdeprecated-declarations]
         traits.set_supports_cool_mode(true);
                                           ^
src/mhi_ac_ctrl.h:354:43: warning: 'void esphome::climate::ClimateTraits::set_supports_heat_mode(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:58): This method is deprecated, use set_supported_modes() instead [-Wdeprecated-declarations]
         traits.set_supports_heat_mode(true);
                                           ^
src/mhi_ac_ctrl.h:355:42: warning: 'void esphome::climate::ClimateTraits::set_supports_dry_mode(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:66): This method is deprecated, use set_supported_modes() instead [-Wdeprecated-declarations]
         traits.set_supports_dry_mode(true);
                                          ^
src/mhi_ac_ctrl.h:356:47: warning: 'void esphome::climate::ClimateTraits::set_supports_fan_only_mode(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:62): This method is deprecated, use set_supported_modes() instead [-Wdeprecated-declarations]
         traits.set_supports_fan_only_mode(true);
                                               ^
src/mhi_ac_ctrl.h:358:39: warning: 'void esphome::climate::ClimateTraits::set_supports_away(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:119): This method is deprecated, use set_supported_presets() instead [-Wdeprecated-declarations]
         traits.set_supports_away(false);
                                       ^
src/mhi_ac_ctrl.h:362:47: warning: 'void esphome::climate::ClimateTraits::set_supports_fan_mode_auto(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:80): This method is deprecated, use set_supported_fan_modes() instead [-Wdeprecated-declarations]
         traits.set_supports_fan_mode_auto(true);
                                               ^
src/mhi_ac_ctrl.h:363:47: warning: 'void esphome::climate::ClimateTraits::set_supports_fan_mode_high(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:86): This method is deprecated, use set_supported_fan_modes() instead [-Wdeprecated-declarations]
         traits.set_supports_fan_mode_high(true);
                                               ^
src/mhi_ac_ctrl.h:364:46: warning: 'void esphome::climate::ClimateTraits::set_supports_fan_mode_low(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:82): This method is deprecated, use set_supported_fan_modes() instead [-Wdeprecated-declarations]
         traits.set_supports_fan_mode_low(true);
                                              ^
src/mhi_ac_ctrl.h:365:49: warning: 'void esphome::climate::ClimateTraits::set_supports_fan_mode_medium(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:84): This method is deprecated, use set_supported_fan_modes() instead [-Wdeprecated-declarations]
         traits.set_supports_fan_mode_medium(true);
                                                 ^
src/mhi_ac_ctrl.h:366:48: warning: 'void esphome::climate::ClimateTraits::set_supports_swing_mode_off(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:131): This method is deprecated, use set_supported_fan_modes() instead [-Wdeprecated-declarations]
         traits.set_supports_swing_mode_off(true);
                                                ^
src/mhi_ac_ctrl.h:367:49: warning: 'void esphome::climate::ClimateTraits::set_supports_swing_mode_both(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:133): This method is deprecated, use set_supported_fan_modes() instead [-Wdeprecated-declarations]
         traits.set_supports_swing_mode_both(true);
                                                 ^
src/mhi_ac_ctrl.h:368:53: warning: 'void esphome::climate::ClimateTraits::set_supports_swing_mode_vertical(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:135): This method is deprecated, use set_supported_fan_modes() instead [-Wdeprecated-declarations]
         traits.set_supports_swing_mode_vertical(true);
                                                     ^
src/mhi_ac_ctrl.h:369:55: warning: 'void esphome::climate::ClimateTraits::set_supports_swing_mode_horizontal(bool)' is deprecated (declared at src/esphome/components/climate/climate_traits.h:137): This method is deprecated, use set_supported_fan_modes() instead [-Wdeprecated-declarations]
         traits.set_supports_swing_mode_horizontal(true);
                                                       ^
Compiling /data/lr_mhi_ac_ctrl/.pioenvs/lr_mhi_ac_ctrl/lib4d9/ESP8266WiFi/CertStoreBearSSL.cpp.o
*** [/data/lr_mhi_ac_ctrl/.pioenvs/lr_mhi_ac_ctrl/src/main.cpp.o] Error 1
========================== [FAILED] Took 3.56 seconds ==========================

closed 2021-06-28 by lifeandfate1

ginkage, 2021-06-28:
Yup, fixed with the latest commit. :) (duplicate of #3 )

lifeandfate1, 2021-06-28:
Yep sorry, totally didn't look before I posted :)
All works now for me, cheers!