ginkage/MHI-AC-Ctrl-ESPHome #7: Odd behaviour ...

swa72, 2021-07-03:
Hi, 
I've tried to get your version running but struggle with an odd behaviour.

If I compile and flash the device OTA with just the core info (ESPHome 1.19.4)

```
esphome:
  name: lr_mhi_ac_ctrl
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: !secret wlan_name
  password: !secret wlan_password

logger:
  level: DEBUG
  baud_rate: 0

api:
  reboot_timeout: 0s

ota:
```
the device connects to WIFI and shows up in HA (but obviously does nothing ;-)). If I add the `includes:` and `climate:` sections the devices flashes correctly and I can ping it. Beyond that, no luck.

```
INFO Reading configuration /config/esphome/lr_mhi_ac_ctrl.yaml...
INFO Starting log output from 192.168.178.44 using esphome API
INFO Connecting to 192.168.178.44:6053 (192.168.178.44)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.178.44: [Errno 111] Connection refused). Re-Trying in 1 seconds
INFO Connecting to 192.168.178.44:6053 (192.168.178.44)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.178.44: [Errno 111] Connection refused). Re-Trying in 1 seconds
INFO Connecting to 192.168.178.44:6053 (192.168.178.44)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.178.44: [Errno 111] Connection refused). Re-Trying in 2 seconds
```

This is the second yaml file

```
esphome:
  name: lr_mhi_ac_ctrl
  platform: ESP8266
  board: d1_mini
  includes:
    - mhi_ac_ctrl.h
    - MHI-AC-Ctrl-core.h
    - MHI-AC-Ctrl-core.cpp

wifi:
  ssid: !secret wlan_name
  password: !secret wlan_password

logger:
  level: DEBUG
  baud_rate: 0

api:
  reboot_timeout: 0s

ota:

climate:
  - platform: custom
    lambda: |-
      auto mhi_ac_ctrl = new MhiAcCtrl();
      App.register_component(mhi_ac_ctrl);
      return {mhi_ac_ctrl};

    climates:
      - name: "MHI-AC-Ctrl"
        id: mhi_ac_ctrl
```

Any idea?

ginkage, 2021-07-03:
Hi!
Yes, long story short, you need to actually plug it into the A/C, otherwise it won't run.
If you just want to kick the tires a bit, you can comment out the lines with "mhi_ac_ctrl_core" in mhi_ac_ctrl.h (e.g. 30, 31, 36-38), and it will start without being plugged in.

swa72, 2021-07-03:
Hi, you da man. Did a test with the commented code - worked. Uncommented and plugged it into the A/C, worked ;-) 
The biggest part was to disassemble and assemble the unit (hint: google for the technical manual). For reference, here is my yaml file that worked with my  SRK25ZS-W, see https://github.com/swa72/home-assistant/blob/main/esphome/roofac.yaml

closed 2021-07-03 by swa72