ginkage/MHI-AC-Ctrl-ESPHome #8: Node not available after flashing

hjkoster400d, 2021-07-08:
Hi, 

First of all, thank you for this easy implementation of the heatpump API using ESPHome. 
I placed the 3 .cpp/.h files in the root of the ESPHome directory of my HA instance, configured the yaml as described. But after flashing the interface returns this message:

[image]

Is this expected or did i do something wrong?

Also, the climate entity does not show up in homeassistant. 
Any idea what this could be?

Thanks for now

ginkage, 2021-07-08:
Hi!
Thank you for the feedback.
Please note that you need to actually plug it into the A/C, otherwise it won't run.
If you just want to kick the tires a bit, you can comment out the lines with "mhi_ac_ctrl_core" in mhi_ac_ctrl.h (e.g. 30, 31, 36-38), and it will start without being plugged in.

hjkoster400d, 2021-07-08:
Ah great, i was just trying to get the software running before plugging it into the AC. Thx for the quick feedback and trouble shooting. So if i'm correct, the climate entity will popup in homeassistant when the wemos is connected to the AC?

ginkage, 2021-07-08:
Yes, it will also show up if you comment the actual control code as I've described above, in case you want to test how it works before you plug it in. Note that you may need to set up ESPHome integration or use HA's discovery feature.

hjkoster400d, 2021-07-08:
Nice, thank you

closed 2021-07-19 by ginkage