ginkage/MHI-AC-Ctrl-ESPHome #11: Multiple units

ogizhelev, 2021-11-20:
Hi there - first thanks for getting this into esphome 
Newbie here - when i flash 2 units both of them start to get random disconnect and become unavailable in esphome .
As far as i understood that might be because of the id , however if i change it it fails to compile .
What changes should i make in order to connect more than 1 unit in the same network ?

ginkage, 2021-11-21:
Hi!
Could you please elaborate on the compilation failure, what error do you get?

ogizhelev, 2021-11-21:
Sure - so in the yaml config if  i change the id like : 
climate:
  - platform: custom
    lambda: |-
      auto mhi_ac_ctrl = new MhiAcCtrl();
      App.register_component(mhi_ac_ctrl);
      return {mhi_ac_ctrl};
    climates:
      **- name: "bedroomac"
        id: bedroomac**

when compiling is trowing those errors : 
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|-- <ESP8266WiFi> 1.0
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESP8266mDNS> 1.2
Compiling /data/mhi-ac-ctrl-bedroom/.pioenvs/mhi-ac-ctrl-bedroom/src/main.cpp.o
/config/esphome/mhi-ac-ctrl-bedroom.yaml: In lambda function:
/config/esphome/mhi-ac-ctrl-bedroom.yaml:83:27: error: 'mhi_ac_ctrl' was not declared in this scope
       return ((MhiAcCtrl*)mhi_ac_ctrl)->get_sensors();
                           ^
/config/esphome/mhi-ac-ctrl-bedroom.yaml: In lambda function:
/config/esphome/mhi-ac-ctrl-bedroom.yaml:101:27: error: 'mhi_ac_ctrl' was not declared in this scope
       return ((MhiAcCtrl*)mhi_ac_ctrl)->get_binary_sensors();
                           ^
/config/esphome/mhi-ac-ctrl-bedroom.yaml: In lambda function:
/config/esphome/mhi-ac-ctrl-bedroom.yaml:84:3: warning: control reaches end of non-void function [-Wreturn-type]
     sensors:
   ^
/config/esphome/mhi-ac-ctrl-bedroom.yaml: In lambda function:
/config/esphome/mhi-ac-ctrl-bedroom.yaml:102:3: warning: control reaches end of non-void function [-Wreturn-type]
     binary_sensors:
   ^
*** [/data/mhi-ac-ctrl-bedroom/.pioenvs/mhi-ac-ctrl-bedroom/src/main.cpp.o] Error 1
========================== [FAILED] Took 3.60 seconds ==========================

ginkage, 2021-11-21:
Ah, got it. Note the following lines in your error output:
```
return ((MhiAcCtrl*)mhi_ac_ctrl)->get_sensors();
return ((MhiAcCtrl*)mhi_ac_ctrl)->get_binary_sensors();
```
These point to the lambdas in `sensor` and `binary_sensor` sections of the yaml file, and are trying to reference your `climate` component — that's where you need to replace `mhi_ac_ctrl` with your id: in this case, `bedroomac`. Note that you can use a more human-readable `name`.

To avoid other conflicts, you'll probably want to change the `name` parameter of the top `esphome` section (second line of the .yaml file), as that's going to be the hostname of your ESP's.

ginkage, 2021-11-27:
Please let me know if the suggestion above helped...

ogizhelev, 2021-11-27:
Hi,
Yeah - whatever i have tried always fails to compile 
here is my full config https://pastebin.com/BDUifscD 
if i make any changes it on the above config it fails.

ginkage, 2021-11-29:
Something like this seems to work fine:

```@@ -80,14 +80,14 @@ climate:
       return {mhi_ac_ctrl};
     climates:
       - name: "Bedroom MHI AC Ctrl"
-        id: mhi_ac_ctrl
+        id: bedroomac
         
         
 # Sensors with general information.
 sensor:
   - platform: custom
     lambda: |-
-      return ((MhiAcCtrl*)mhi_ac_ctrl)->get_sensors();
+      return ((MhiAcCtrl*)bedroomac)->get_sensors();
     sensors:
       - name: "Error code bedroom"
       - name: "Outdoor temperature bedroom"
@@ -129,7 +129,7 @@ sensor:
 binary_sensor:
   - platform: custom
     lambda: |-
-      return ((MhiAcCtrl*)mhi_ac_ctrl)->get_binary_sensors();
+      return ((MhiAcCtrl*)bedroomac)->get_binary_sensors();
     binary_sensors:
       - name: "Defrost bedroom"
 
```

ogizhelev, 2022-01-30:
That worked

closed 2022-01-30 by ogizhelev