ginkage/MHI-AC-Ctrl-ESPHome #12: External room temperature

serrj-sv, 2021-11-30:
Hi!
There is a new feature in original project: send temperature data to AC unit from external sensor:
https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/SW-Configuration.md#room-temperature

Is it possible to port it here? Thanks!

ginkage, 2021-11-30:
Sounds like a good idea, I'll try to implement that when I have some spare time. The ability to reference other entities' attributes (added to ESPHome a couple months ago) should come in quite handy.

synegic, 2022-04-18:
I've added this 'feature' in my pull request; https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/pull/16. If you want, you could test it out and let me know if it works for you.

closed 2022-04-23 by ginkage