ginkage/MHI-AC-Ctrl-ESPHome #13: State change towards Home Assistant

Siumba, 2022-02-09:
Something strange is going on when letting the device sit idle. It changes to Unavailable and a couple of seconds later the reported state is that it's one with a specific HVAC_mode. However, the actual device itself stays turned off, it's only the Wemos D1 mini chip that updates it's state.
[image]

EleeeD, 2022-03-22:
I also experience issues with wrong states and power on/off behaviour. 

Have a split MHI AC with this MHI-AC-CTRL setup and -module in each indoor unit. 
Latest versions of Home Assistant with Esphome on a Raspberry Pi. Also running the Homekit integration on Homeassistant in order to control the ACs from iOS.  I don't have any automations configured. 

- Recently had an unexpected power outage and when power was switched on again, the AC switched on as well, although it was off before the power outage. 
- In order to reproduce and look for possible cause of the above, I tested by manually powering off both the AC and Pi and again powering them on, and this time the ACs showed an active state (heating) in both Homeassistant and iOS Home, although the ACs were and remained idle. 
- After unplugging/rebooting the Pi only, the state is displayed correctly as off.

synegic, 2022-04-18:
This should be fixed in my pull request; https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/pull/16. If you want, you could test it out and let me know if it works any better.

closed 2022-04-23 by ginkage