ginkage/MHI-AC-Ctrl-ESPHome #29: mhi_ac_ctrl_core.loop error: -4

Koendejongh, 2022-07-29:
Hi, I just installed this device, I have no airco readings. 
The Wemos d1 is connected to HA (all diagnostic values are there).

In the esphome log it continuously shows:  mhi_ac_ctrl_core.loop error: -4

Any clue?

Thanks,
Koen

zuohaot, 2022-08-21:
I also saw mhi_ac_ctrl_core.loop error: -2 and mhi_ac_ctrl_core.loop error: -1 in the log as well.

extera-nl, 2022-09-01:
I have the same error

[00:07:54][W][mhi_ac_ctrl:077]: mhi_ac_ctrl_core.loop error: -4

Would be great is this can be fixed. I have 4 working AC units now, flashed with older version op ESPhome, but can get a new unit connected now.

Let me know if I can test something

Koendejongh, 2022-11-05:
Hi all,

I fully reassembled the hardware and now it works. No more loop errors!

schol478, 2022-11-05:
The hardware was the problem for me also!

zuohaot, 2022-11-06:
> The hardware was the problem for me also!

What was causing problems? All my 7 devices are reporting loop error: -2

gabcraftia, 2022-12-10:
for me it is solved when i plugged it into my AC. When i powered it via micro usb it was giving me error: -4

MrAlfabet, 2023-01-27:
I'm getting this error on all 3 of my devices. Anyone else with the latest version?

melmager, 2023-07-24:
mhi_ac_ctrl_core.loop error: -4 - was for me missing clk connection ; Hardware: D1 Mini and own PCB
now loop error -1 , But on / off AC works 
I think -1 is Timing Problem -

RobertJansen1, 2024-01-25:
Seems like this is hardware related, if there are issues you think can be solved in software please reopen

closed 2024-01-25 by RobertJansen1

petruknw, 2025-05-24:
Is this compatible with the hardware that it was based off https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/Hardware.md ? Or have I found out the hard way it's not? I'm using the hardware in my link, but even after soldering a second PCB, I still have the same error.