ginkage/MHI-AC-Ctrl-ESPHome #30: Power Status always on on reboot

aferrato, 2022-08-11:
Hello,
when the esp8266 is rebooted, the status of the air conditioner changes to on with the last mode used even if it is actually off.

I have tried what is written here  #2, but it doesn't work.

To solve I added this code:

```
switch (this-> power_) {
     case 0:
         this-> mode = climate :: CLIMATE_MODE_OFF;
         this-> publish_state ();
         break;
     default:
         break;
 }
```

inside `void cbiStatusFunction (ACStatus status, int value) override` before `switch (status)`

ervee, 2022-08-15:
I have this too. Will give your solution a try.

sanderlv, 2022-10-06:
Like this works indeed!
[image]

ginkage, 2022-10-06:
I've committed this workaround: https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/commit/a43bc7187dba3ea520b662487ad94af0bdf9fdfb
Please check if it works correctly.

sanderlv, 2022-10-08:
I can confirm it works, thanks!

closed 2024-08-30 by RobertJansen1