ginkage/MHI-AC-Ctrl-ESPHome #32: Strange temperature readings when at least one of my ACs is turned ON

zuohaot, 2022-08-21:
**Home Assistant Setup**
Home Assistant: 2022.8.5
Supervisor: 2022.08.3
Operating System: 8.4
ESPHome: 2022.8.0

**MHI-AC-Ctrl Setup**
Hardware: D1 mini
External temperature probe used?: No
Firmware: MHI-AC-Ctrl-ESPHome

**AC Setup**
- 4 * SRK35ZMXA-S
- 3 * SRR35ZM-S

**Problem Description**
I'm in Australia where is in winter atm. When I turn on one or more ACs, the room temperature reading of almost all the other ACs (which are OFF) will rise up dramatically. The 2nd snapshot was captured around 15 minutes after I turned on the "Office AC" on the ground floor. As you find, the temperature readings of all the ACs of first floor are about 10-15 dgrees higher than the actual room temperature. I wonder if this issue is caused by mhi-ac-ctrl or the AC indoor unit itself? TIA!!!

TimCinel, 2022-08-21:
Im experiencing the same issue and am unsure how to determine if it’s an indoor unit issue or MHI-AC-Ctl issue.

FWIW, I do have 7 other units not presenting the same problem.

Multi head system, specs:
SRK25ZMXA-S Indoor
SCM60ZM-S1 Outdoor

Klapperman, 2022-09-06:
Are these units part of a single multi-split unit? Because in that case this can be explained as the units themselves will warm up slightly when the central pump starts heating the lines for the one unit you turned on. And because the fans are not running the internal temp of the unit will rise (these tempts are very similar at my home in the same use case).

Holewijn, 2022-12-09:
I have the same issue here.

WorlockM, 2023-02-09:
I have 2 dual unit ACs. So 2 outdour units and 4 indoor units. I have the same problem, but only on the AC that is connected to the same outdoor unit. But I noticed that it's not only the temperature reading of the AC, my netatmo sensor also shows that the room really is getting warmer when the AC in the other room is turned on.

closed 2023-03-14 by zuohaot