ginkage/MHI-AC-Ctrl-ESPHome #33: Restore previous settings from hassio

perrykipkerrie, 2022-08-25:
First thank you for this nice port of the software, it works almost flawlessly for me.
However, sometimes the esphome restarts (once a day or something, due to unknown reasons) and the HAVC is turned off (due to the default off setting in the code).

Do you know if there is a possibility to read out the current value exposed by home assistant?, as you do not want to default the HAVC to cooling (as restarting the system will consequently turn on the unit.)

ervee, 2022-08-30:
I'm not sure if this works for your case. Just passing by trying to help. But did you see https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/issues/30#issue-1335529718 and perhaps if that works for you?

perrykipkerrie, 2022-08-30:
Thank you for passing by :)

The problem is actually the other way around, in #30 the status is the one provided by home assistant, while it is supposed to be off. However, in my case, when it reboots it is powered off, while it should be powered on (it should get the state from the hassio instance). 

I will mess around with the code soon to see if i can implement this, defaulting to off seems in most cases a safe consideration.

synegic, 2022-10-09:
It's not really reliable to try to get/save power status from HASS or any external source. It may cause weird behaviour where AC's will spontaneously turn on while they in fact should be off. I think the default of power off is the best.

You could probably create a sync automation which checks whether the state from HA mismatches the one in the unit, and then apply the requested mode by setting the mode if it does.

closed 2024-08-30 by RobertJansen1