ginkage/MHI-AC-Ctrl-ESPHome #34: Forked, but up-to-date?

sanderlv, 2022-09-19:
Hi,  I read:
MHI-AC-Ctrl-core.* files were forked directly, with no modification

Does this mean the original files from below are not needed to be updated?
https://github.com/absalom-muc/MHI-AC-Ctrl/tree/master/src

jeroenkl, 2022-09-25:
@ginkage is it?

ginkage, 2022-09-25:
Even with "normal" GitHub forks (which this project isn't, as it only uses a handful of files), the original files would have to be updated manually. At the moment, the files are slightly behind the original repo.
Unfortunately, due to the war in Ukraine, I don't have access to my AC unit, and won't be able to test anything for a while...

ginkage, 2022-10-08:
I mean, try and see if it works (and create a pull request if it does), by all means. :)

closed 2024-01-25 by RobertJansen1