ginkage/MHI-AC-Ctrl-ESPHome #35: Climate degrees steps

sanderlv, 2022-10-06:
Hi,

is it possible to set the climate entity temperature in half degrees?

synegic, 2022-10-09:
Hello, this seems to be a limitation in the used library (MHI-AC-Ctrl) as it only accepts unsigned integers, not float values. IIRC there was a discussion about this on that repo as well. If they can do it in the main library, we can make it happen in the ESPHome variant. In the meantime, I'd prefer not to break library parity/compatibility.

sanderlv, 2022-10-09:
Ok, thats a pity

forerunnert, 2022-11-17:
In the [documentation](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/SW-Configuration.md) it says:

_Target room temperature (float) in °C, resolution is 0.5°C_

for topic Tsetpoint.

closed 2022-12-14 by sanderlv

serrj-sv, 2023-08-04:
> In the [documentation](https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/SW-Configuration.md) it says:
> 
> _Target room temperature (float) in °C, resolution is 0.5°C_
> 
> for topic Tsetpoint.

fix: https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/pull/62