ginkage/MHI-AC-Ctrl-ESPHome #36: Low, Medium, High, Auto

sanderlv, 2022-10-08:
Hi,

Low, Medium, High seems to do what they should, at least, it starts low, then medium, then high. But the auto blows all paper from the desk and does not really seem to be going "down" or "up", just at boost speed.

Why?

sanderlv, 2022-10-08:
Ah... it seems a known issue.

https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/issues/22

Not sure what the best way forward is now...?

sanderlv, 2022-10-08:
It seems "some" progress is made, see last comment here:
https://github.com/absalom-muc/MHI-AC-Ctrl/issues/99

Is this something that can be added in this repo possibly?

ginkage, 2022-10-08:
Feel free to try and pull the latest 2.6R2 sources from the original repo and map the (finally implemented) proper "Auto" mode.
Unfortunately, due to the war in Ukraine, I don't have access to my AC unit, and won't be able to build and/or test anything for a while...

synegic, 2022-10-08:
I will take a look at this and all other open requests tomorrow. See if I can make some of the requested changes.

synegic, 2022-10-09:
Pull request has been opened.

PS @ginkage I really hope you can go home again soon and that the madness of what's happening in Ukraine will finally end.

sanderlv, 2022-10-09:
I also hope for the best in the Ukraine situation!

Please let me know when things are "done" as described above and I will test.

(I'm a user, not a coder :-))

Thanks!

synegic, 2022-10-09:
@sanderlv my changes have been merged into master, so you can test with the latest code in the repo (just copy over the new files like you did when 'installing' it)

sanderlv, 2022-10-09:
Ah cool, and may I ask what has changed? In normal (not coding) language?

synegic, 2022-10-09:
You can find some more details about it here; https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/pull/38.

sanderlv, 2022-10-09:
Ok I updated all. currently too hot to heat so I can't test the speeds.

Waht is "vanes" for? Is that something I should/can set from home assitant? Or is it only to be read?

sanderlv, 2022-10-09:
Never mind, got it.

synegic, 2022-10-09:
Vanes represent the direction of the air output from the unit. It has a couple of values; 1 through 5. You can try it yourself by setting the value from an automation in home assistant. Simply choose 'Call service' as action, and search for 'vanes'. It will show up in the autocomplete list.

Set one of the values (1-5) to see what it does on your unit. Basically it will allow you to change the direction of the air coming out of the unit, like with the remote.

sanderlv, 2022-10-09:
Got it! And love it!

sanderlv, 2022-10-09:
One other thing... the swing mode is "up/down"? 
how can I control the "left/right" (black) swing?

synegic, 2022-10-09:
Sadly that is technically not possible to be set through the MH-AC-Ctrl library/SPI interface. Only vertical swing can be chosen.

ervee, 2022-10-09:
Is it? I have seen screenshots with L/R vane control in the new App with the new Wi-Fi module. It seems obly the protocol needs to be decoded. Or did someone try and fail in this?

synegic, 2022-10-09:
The only things we can currently do are the things the core control library support/are reverse engineered. This reverse engineering has to be done by people with the WiFi module handy. Communication can then be sniffed.

As long as no-one steps forward with the required technical sniffed packets/capture of the data send by the app, it will not be supported. https://github.com/absalom-muc/MHI-AC-Ctrl/issues/77

ervee, 2022-10-15:
I think this can be closed. With the latest pull from synegic it works pretty well within the limits of ESPHome.

closed 2022-12-14 by sanderlv