ginkage/MHI-AC-Ctrl-ESPHome #37: Power to kWh

sanderlv, 2022-10-08:
Is it possible to have a KwH sensor like below:
```
sensor:
  - platform: uptime
    name: ${devicename} Uptime
  - platform: wifi_signal
    name: ${devicename} WiFi Signal
    update_interval: 30s
  - platform: custom
    lambda: |-
      return ((MhiAcCtrl*)id(${deviceid}))->get_sensors();
    # Sensor names in HA, you can change these if you want
    # Don't delete them or change their position in the list
    sensors:
      - name: ${devicename} - Error Code
      - name: ${devicename} - Outdoor Temperature
      - name: ${devicename} - Return Air Temperature
      - name: ${devicename} - Outdoor Unit Fan Speed
      - name: ${devicename} - Indoor Unit Fan Speed
      - name: ${devicename} - Current Power
        id: power
      - name: ${devicename} - Compressor Frequency
      - name: ${devicename} - Indoor Unit Total Run Time
      - name: ${devicename} - Compressor Total Run Time

  - platform: total_daily_energy
    name: "${devicename} - Dagverbruik"
    power_id: "power"
    filters:
        # Multiplication factor from W to kW is 0.001
        - multiply: 0.001
        - throttle: 300s
    unit_of_measurement: kWh
    icon: mdi:clock-alert
```

I do not seem to get some output via "id:power"

synegic, 2022-10-08:
That 'power' attribute is not energy (Kwh like you're after) but current. So amperage. What you want to do is technically impossible. You could however set up some complex formula that also takes into consideration the actual voltage and then calculate the watts, poll that value and take the average to sum up as Kwh, but it's just not going to be anywhere near reliable this way. Better to just insert a CT clamp/Kwh meter between the unit and your mains power.

sanderlv, 2022-10-09:
Thank you, and yes, stupid A is not W, I will make that P=U*I :-).

But apart from that, will the "id:power" in esphome provide me the value? to be usable in templates etc.?
And why is is even called "power" if it Amps :-), shouldnt it be "Amperage"?

synegic, 2022-10-09:
Very valid point. I guess I doubted there a bit when I refactored the sensor name(s)/id(s). The library we use calls it 'power' internally, I guess I made 'current power' of that just to keep a reference for the people that used the library natively before.

But yes, current (so not current power) would be the more accurate label/sensor ID.

synegic, 2022-10-09:
Actually, someone already posted a pretty complete example here as well; https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/pull/21#issuecomment-1160188210.

sanderlv, 2022-10-09:
Got it like this:

```
sensor:
  - platform: custom
    lambda: |-
      return ((MhiAcCtrl*)id(${deviceid}))->get_sensors();
    # Sensor names in HA, you can change these if you want
    # Don't delete them or change their position in the list
    sensors:
      - name: ${devicename} - Error Code
      - name: ${devicename} - Outdoor Temperature
      - name: ${devicename} - Return Air Temperature
      - name: ${devicename} - Outdoor Unit Fan Speed
      - name: ${devicename} - Indoor Unit Fan Speed
      - name: ${devicename} - Current Amperage
        id: ampere
      - name: ${devicename} - Compressor Frequency
      - name: ${devicename} - Indoor Unit Total Run Time
      - name: ${devicename} - Compressor Total Run Time
    
  - platform: template
    name: ${devicename} - Current Power
    id: "power"
    unit_of_measurement: W
    lambda: return id(ampere).state * 230;

  - platform: total_daily_energy
    name: "${devicename} - Dagverbruik"
    power_id: "power"
    filters:
        # Multiplication factor from W to kW is 0.001
        - multiply: 0.001
        - throttle: 300s
    unit_of_measurement: kWh
    icon: mdi:clock-alert
```

[image]

It does it for now. To get an "idea". Later I will import the actual voltage but for now this is fine.

thanks!

closed 2022-12-14 by sanderlv

nashy008, 2023-11-29:
Is this Amp reading accurate or only the current for inside unit? It seem lower than I expected.

ervee, 2023-11-29:
I have not measured the amps with a clamp or something but it does look right for me.

[image]

It is the current for the outside unit. Or perhaps better said, the current for the entire system because the inside units are powered from the outside unit. I have an SCM60ZS-W outdoor unit that can pull 5.9 Amps max (in heating mode) according to the spec sheet and it seems to be getting close to it.