ginkage/MHI-AC-Ctrl-ESPHome #41: Room Temperature

JoepWi, 2022-10-11:
Is there a way to replace the room-temperature (return air temperature) with an external EvoHome sensor available in HA value to steer the units settings.
In my experience there is rather a big difference between the two where the EvoHome one is more accurate.

ginkage, 2022-10-11:
Should be possible: the original sources have `MHI_AC_Ctrl_Core::set_troom()` method specifically for that purpose.

synegic, 2022-10-11:
Yes, this is actually already implemented. Call the service `set_room_temperature` from HA automations to give the unit a value from another sensor.

https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/blob/master/lr_mhi_ac_ctrl.yaml#L52

JoepWi, 2022-10-11:
Should I make an automation to update that regular or can I just replace it in some smart template?

fonske, 2022-10-11:
@JoepWi 
Thats correct

- id: '1663182606294'
  alias: MHI Temp overule
  description: ''
  trigger:
  - platform: time_pattern
    seconds: '59'
  condition: []
  action:
  - service: esphome.mhi_woonkamer_set_api_room_temperature
    data:
      value: '{{ states(''sensor.netatmo_woonkamer_temperature'') }}'

ervee, 2022-10-15:
I tried this but my AC seems to be heating well over its set point when using the external temperature api. It looks like the AC is expecting the temperature sensor to be a couple of degrees higher because it is normally on top of the unit and hot air rises and compensates for this. But overcompensates with the external temperature sensor.

synegic, 2022-10-15:
@ervee I think this is normal. The AC wants to be sure the setpoint is reached so it will spend some more time heating. It will prevent 'pendeling' which means turning on/off all the time. The same behaviour is seen when using only the 'return air' or internal sensor. It will also stay heating a long time after it reached the setpoint. I guess you can/should compensate for it through your API call automations.

ervee, 2022-10-15:
Yeah perhaps I'll give it another try when we need daily heating to get a good feel for it.

Anyway, I think this issue can be closed!

Darosnl, 2022-10-17:
> @JoepWi Thats correct. In automations.yaml
> 
> * id: '1663182606294'
>   alias: MHI Temp overule
>   description: ''
>   trigger:
>   
>   * platform: time_pattern
>     seconds: '59'
>     condition: []
>     action:
>   * service: esphome.mhi_ac_ctrl_set_api_room_temperature
>     data:
>     value: '{{ states(''sensor.netatmo_woonkamer_temperature'') }}'
> 
> Remark: the dots should be"-" in the code

Thanks.
But i have 4 ac's here with ESPHome, i could not find where to specify wich AC its needs to override.

fonske, 2022-10-17:
@Darosnl 
service: esphome.**mhi_ac_ctrl**_set_api_room_temperature

mhi_ac_ctrl is the name of the airco unit, you have 4 different names i pressume?

Darosnl, 2022-10-17:
> @Darosnl service: esphome.**mhi_ac_ctrl**_set_api_room_temperature
> 
> mhi_ac_ctrl is the name of the airco unit, you have 4 different names i presume?

Thanks and thanks for the quick reply!

Marijn1250, 2022-12-18:
In the automation I added a +2 to the garage temp. So if it really is 16 degrees, it tells the AC it is 18 degrees.  
The AC will throttle down to hold the lowel temp. 

service: esphome.airco_garage_set_api_room_temperature
data:
  value: "{{ states('sensor.s20_garage_ds18b20_temperature') | int +2}}"

ervee, 2023-02-22:
This works, but now the AC is reporting the +2 temperature back to Home Assistant so the temperature card now also shows the +2 temperature 😒

I was testing another solution by editing the mhi_ac_ctrl.h file. In the void set_room_temperature I just added another 8 (units are 0.25 C steps I presumed) to the set_troom value. But the reported temperature back to HA is the same +2C.

So I tried putting an if then part to the "case status_troom" part where I think the AC reports back the temperature to HA. If the api timer is not expired that it would report back the temperature -2C but I did not get that working.

I'll keep trying, but perhaps someone else has a better solution?

Marijn1250, 2023-03-09:
My workaround only works with an external temp sensor. It is manipulating the temp given to the AC. 
I actually don't care what temperature it shows, when I am cold, I simply set it higher and vice versa. 
 
Maybe you can ignore the AC thermostat on the dashboard completely and create your own using templates. 
This was you could manipulate the original thermostat and look at your own creation with the actual temp. 

The room temp is always different to what the AC tells. That sucks in warm air from the sealing. It can be 15 degrees in my room while the AC is showing that it is 18 at ceiling height. So i find that useless anyway.