ginkage/MHI-AC-Ctrl-ESPHome #42: Total run time is, was and stays 100h

sanderlv, 2022-10-13:
Why is my total runtime 100 since installation and keeps a 100h?

[image]

ervee, 2022-10-15:
I also seem to have this issue.
It did work a while ago I'm pretty sure.

Could it be that there's something wrong with the sensor code at the bottom of mhi_ac_ctrl.h together with the sensor lines in the yaml file? It looks like they don't match. But I don't know if they should exactly match. 

[image]

synegic, 2022-10-15:
@ervee A missing sensor in your YAML (for example, vanes position which was added in latest version) should not really affect anything as long as the newly added sensor is last in the list. If you do not specify a matching index in your YAML, it will not do anything.

Now my question would be; does it actually report it back correctly with the latest MHI-AC-Ctrl firmware? I've checked the code and it still uses the same calculations as the original MHI-AC-Ctrl, we also haven't really changed anything related to this.

ervee, 2022-10-15:
If I connect to the console via ESPHome or web interface I don't see it reporting these sensors/values like it does with uptime, temperatures and wifi status. But I'm also not sure it should or this value is polled or because the run time is currently not changing.

ervee, 2022-10-15:
This is what I ment with the sensor code:

This is mhi_ac_ctrl.h:
[image]

This is the yaml:
[image]

Should the binary sensor defrost at line 463 of the .h file not go below the vanes sensor line or does this order not matter? I fail to see the logic of the mapping of sensors in the .h file and the yaml and if order matters here.

sanderlv, 2022-10-16:
I am very curious. If I can do something to trouble shoot let me know.

JoepWi, 2022-10-16:
I see the same with my units. Also 100H.

DYLaKo, 2022-10-17:
Mine is at 500h (AC unit on the first floor), with the MHI-AC-Ctrl-ESPHome software from a few months ago. After updating to the latest version available, it still shows 500h. This unit is not used alot.

As far as I remember (and also if I check with my AC in the living room), the value for this sensor is only updated every 100 hours of runtime. For example, the sensor of my living room AC (which is running the MHI-AC-Ctrl software, not the ESPHome version) has a total runtime of 4100 hours, also rounded on 100 hours. This AC unit is running for cooling in summer and heating in winter. 

If you are sure your AC has run more than 200 hours (which should trigger it to jump to 200h), then something could be wrong with the sensor. But if you expect it to change every hour, I hate to disappoint you, that won't happen since the AC is not giving this data every running hour.

ervee, 2022-10-17:
Ah, that would make sense for my counters at least.

ervee, 2022-11-19:
And recently it updated so all is ok. How is it for you @sanderlv ?

[image]

sanderlv, 2022-11-19:
Still at 100h and it ran a lot... but not sure if it was 100h