ginkage/MHI-AC-Ctrl-ESPHome #43: ESP32 - D1 Mini Pro

sanderlv, 2022-10-27:
Hi, I have tried to use a D1 Mini Pro ESP32 instead of an ESP8266. But to no avail, it just wont boot.

I want to use the advantage of also using the new bluetooth proxy component in the ESP32.

Is there any hard coding towards ESP8266?

Is it possible to use an ESP32?

synegic, 2022-10-28:
First thing that comes to mind is that you should check the pins. The pin layout is different, and you may need to change some around in the core library's h and/or c files. Also, I think you can remove the CPU frequency option as it may be invalid for ESP32.

sanderlv, 2022-10-28:
I think, that the pin layout is pretty equal. The other tip I will try somewhere next weeks.

And about the other, I hope someone can confirm/deny that some programmatic changes are needed (not yaml, but the others).

synegic, 2022-10-28:
If you can't even get it to boot/join your WiFi, something is probably trying to do something that is not possible or causes an error on the ESP32. You might be able to hook-up the ESP32 to the serial monitor built-into the Arduino desktop app to debug a bit what error is triggered.

sanderlv, 2022-10-28:
I think from now on ESP32 is the only viable option and most future proof. Lower power and better wifi and BLE.

I really would appreciate support for this 😀

synegic, 2022-10-30:
If you can test it out with above pinout and it works, the maintainer of the library could consider the possibility of setting the pins in such a way that it remains drop-in compatible with this project.

synegic, 2022-11-08:
Have you tested it out yet with above config? I also got intrigued by the possibilities of using an ESP32 instead of an 8266 and eventually extending the module in the AC with (for example) a PM2.5/PM10 sensor. So I've also ordered some ESP32 Mini's now.

sanderlv, 2022-11-08:
Hi, no not yet and no active plans. Since I discovered shelly plus which have esp32 and I can flash with esphome I do not really need them yet...

sanderlv, 2022-11-19:
> 

Can you please elaborate on "how to submit" a feature request? Still would like to have ESP32's in my aircos...

ElBarto333, 2023-01-13:
Little update for everybody else who might want to try to use an ESP32 D1 Mini.
I got it running by changing the pin definitions in [MHI-AC-Ctrl-core.h](https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/blob/master/MHI-AC-Ctrl-core.h) as pointed out by @synegic.
```
#define SCK_PIN  18
#define MOSI_PIN 23
#define MISO_PIN 19
```
My config in ESPHome looks like this:
```
esphome:
  name: some-name
  platform: ESP32
  board: esp32dev
  platformio_options:
    # Run CPU at 160Mhz to fix mhi_ac_ctrl_core.loop error: -2
    board_build.f_cpu: 160000000L
  includes:
    - mhi_ac_ctrl.h
    - MHI-AC-Ctrl-core.h
    - MHI-AC-Ctrl-core.cpp
```
Everything else is unchanged.

When the ESP32 is only connected via USB to my PC I get the same error as in #29. 
But as soon as I plug it into my indoor unit I can control my air conditioner via ESPHome/Home Assistant.

Hope this helps.

acfnews, 2023-01-15:
Hi all, I also like to help with this (as I would really like to use WPA3 WiFi features only available on ESP32), but to enlarge the success factor, I am wondering where you guys but your ESP32 boards, and what version?
I am thinking on buying 5 pieces at once (to have some spare and for fiddling around).

I only have 2 TTGO devices left, which do not suffice for the MHI interface.

Any tips for decent price versus normal shipping times in Europe are greatly appreciated.
I did find for example: 
[amazon](https://www.amazon.nl/AZDelivery-Microcontroller-ESP32-WROOM-32-compatibel-inclusief/dp/B08BTRQNB3/ref=asc_df_B08BTLYSTM/?tag=nlshogostdde-21&linkCode=df0&hvadid=454898017909&hvpos=&hvnetw=g&hvrand=8881571982427953346&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=2528&hvtargid=pla-956539855876&th=1)

I just want to buy the proper board so I can focus on the MHI side of things.

ervee, 2023-01-15:
Last time I bought an ESP it was from TinyTronics. I bought this one:
https://www.tinytronics.nl/shop/nl/development-boards/microcontroller-boards/met-wi-fi/wemos-c3-mini-esp32-c3

Mind you this is not a D1 but C3 variant with the RiscV processor architecture and I'm not sure if it does WPA3. They have other variants too.

I also order from HobbyElectronica:
https://www.hobbyelectronica.nl/product-categorie/wemos/boards-wemos/

I don't know too much about the success factor thought. I think the D1 mini V3 is pretty okay but heard people who order from sketchy sources getting bad clones with mixed results.

hberntsen, 2023-01-16:
I've got an Ai-Thinker ESP-C3-32S-Kit board. I'm rewriting the code to use hardware SPI on the IDF platform. So harware-wise that's possible

ervee, 2023-01-16:
That's nice!
I also got an ESP32 C3 13U 2MB board a while ago . It was cheap but it does not have a wifi antenna on the board (you need an antenna with a wire) so keep that in mind. Also the small flash size is a pain but I got my project running on it with OTA capability but I'm pretty sure the MHI project is way too big for this board so...

glsf91, 2023-02-07:
> I've got an Ai-Thinker ESP-C3-32S-Kit board. I'm rewriting the code to use hardware SPI on the IDF platform, which works fine :). So harware-wise that's possible

Did you had any success with this?
Are still using esphome with this modified code?

hberntsen, 2023-02-07:
Yes, I've got two running in my home right now, works great. I'm not on full feature parity with this repo yet but works fine so far. I'm looking at the developments in https://github.com/absalom-muc/MHI-AC-Ctrl/issues/77 to implement those as well.

@glsf91 are you interested? My code needs some cleaning but I can upload whatever I already have if that helps

glsf91, 2023-02-07:
@hberntsen 
Yes, I'm interested. If you can upload your code somewhere (github?) that would be nice.
I already used an esp32 with MHI-AC-Ctrl code (not esphome) but experience too much mhi_ac_ctrl_core.loop errors. I managed to make a better but I'm not satisfied yet. Have also to take a look again.
So I was interested if IDF and hw SPI was doing a lot better.

hberntsen, 2023-02-07:
I'm also tracking Bluetooth advertisements on both ESPs (for temperature), the one with 8 temperature sensors attached to it sometimes generates frame errors:
[image]

The other one with just one sensor a lot less (0 since last boot). I did not investigate the cause of them. Given that we receive a new frame every 40ms the error rate is quite low :). I'm planning to report the Bluetooth temperature to the AC as reference temp.

The trick with Hardware SPI on my ESP32-C3 was to generate a CS signal on the ESP itself, and loop it back to the hardware SPI CS pin. Otherwise It would never receive a frame. The ESP docs say that it should be able to operate without CS signal but I could not get that to work at all.

I've dumped my code at https://github.com/hberntsen/mhi-ac-ctrl-esp32-c3 :)

glsf91, 2023-02-07:
A lot of work to change this. I'm not familiar with IDF, so takes some time to try to understand your code. 
I also want tracking Bluetooth advertisements.

> The trick with Hardware SPI on my ESP32-C3 was to generate a CS signal on the ESP itself, and loop it back to the hardware SPI CS pin. Otherwise It would never receive a frame. The ESP docs say that it should be able to operate without CS signal but I could not get that to work at all.

Strange that the other project you referring to (homekit) is not using a CS.

hberntsen, 2023-02-08:
> Strange that the other project you referring to (homekit) is not using a CS.

Indeed, though that project is targeting the ESP32, not the C3 variant.

glsf91, 2023-02-08:
I have an ESP32 (D1 mini ESP32 clone). So this means it maybe could work without CS and the mhi-ac-ctrl PCB.
I will give it a try later. First trying some other stuff.

thehaxxa, 2023-08-20:
@ElBarto333
> Little update for everybody else who might want to try to use an ESP32 D1 Mini.

Can anyone comment on the long term stability, I am considering using a ESP32 as it's a lot better option in most contexts in 2023. 

Did you encounter any issues? Apart from setting the pin definitions and CPU frequency, did everything else simply 'work'?

ElBarto333, 2023-08-20:
@thehaxxa 
> Can anyone comment on the long term stability, I am considering using a ESP32 as it's a lot better option in most contexts in 2023.
I have ESP32s and and 8266 running in my units. None of them had any issues until now.

HpNoTiQ56, 2024-05-20:
Hi,
I've switch to ESP32 over 8266 to get BT proxies. I had no error before so it's not an hardware problem.
I've got lots of [mhi_ac_ctrl:141]: mhi_ac_ctrl_core.loop error: -1 and -4 with this code : 

`# Version 2.1
substitutions:
  # Unique device ID in HA
  deviceid: "XXX"
  # Unique device name in HA (sensor names will be prefixed by this name)
  devicename: "Climatisation salon"
  adress_ip: XXX

esphome:
  name: ${deviceid}
  friendly_name: XXX
  #platformio_options:
    # Run CPU at 160Mhz to fix mhi_ac_ctrl_core.loop error: -2
  #  board_build.f_cpu: 240000000L
  includes:
    - mhi_ac_ctrl.h
    - MHI-AC-Ctrl-core.h
    - MHI-AC-Ctrl-core.cpp

esp32:
  board: wemos_d1_mini32
  framework:
    type: arduino

ota:
  password: ""

packages:
  # Comment if you encounter mhi_ac_ctrl_core.loop error: -2 errors and uncomment the legacy_framesize file
  mhi-ac-ctrl: !include conf/large_framesize.yaml
  # Uncomment this to allow for a working configuration on floor units or older units
  # mhi-ac-ctrl: !include conf/legacy_framesize.yaml

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
  

  
# Optional manual IP
  manual_ip:
    static_ip: ${adress_ip}
    gateway: 
    subnet:

  ap:
    ssid: ${devicename}

captive_portal:

# Enable bluetooth tracker/proxy

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: false

bluetooth_proxy:
  active: true

logger:
  level: DEBUG
  baud_rate: 0
  logs:
    component: ERROR

button:
  - platform: restart
    name: Restart
    entity_category: diagnostic

api:
  encryption:
    key: 

  reboot_timeout: 0s
  services:
    # Call the set_api_room_temperature service from HA to override the room temperature
    # If a new value has not been received after room_temp_api_timeout seconds, it will fall back to internal sensor
    - service: set_api_room_temperature
      variables:
        value: float
      then:
        - lambda: |-
            return ((MhiAcCtrl*)id(${deviceid}))->set_room_temperature(value);
    # Call the set_vanes service from HA to set the vane position
    # Needed because the ESPHome Climate class does not support this natively
    # Possible values: 1-4: static positions, 5: swing, 0: unknown
    - service: set_vanes
      variables:
        value: int
      then:
        - lambda: |-
            return ((MhiAcCtrl*)id(${deviceid}))->set_vanes(value);

sensor:
  - platform: uptime
    name: Uptime
  - platform: wifi_signal
    name: WiFi Signal
    update_interval: 60s

text_sensor:
  - platform: version
    name: ESPHome Version
  - platform: wifi_info
    ip_address:
      name: IP
    ssid:
      name: SSID
    bssid:
      name: BSSID`
Do anyone using esp 32 got rid of these errors?

swerveshot, 2024-10-19:
> Little update for everybody else who might want to try to use an ESP32 D1 Mini. I got it running by changing the pin definitions in [...]

I'm really interested in running this project on an ESP32 as well. Like many I think having the BT proxy feature is what makes this controller the better option over the ESP8266 it was originally created for.

But now for a complete n00b question; how should one make these modifications? Simply fork the project, make the modifications and install from your own repo like this?

```
external_components:
  - source: github://myuser/MHI-AC-Ctrl-ESPHome@master
    components: [ MhiAcCtrl ]
```

And when doing so, how would one keep track of changes in the main project. If after all you are simply looking for a few changes in the PIN assignment?

Scales82, 2025-09-14:
Im interested in doing this also to extend my bluetooth around the home. Im running the Fonske SMD boards for my MHI ctrl right now. What ESP32 device would be the best one to 'slot in' the existing 8266 pin headers I already have? looking for a pure 'flash n swap' solution with the yaml

RobertJansen1, 2025-09-14:
For a 1-on-1 replacement of a d1 mini, I would stick to https://www.wemos.cc/en/latest/c3/c3_mini.html for now. in the future, i think i would prefer the c6 with 2.4Ghz wifi 6 capabilities i think, but with esp32 there is a lot of things to choose from and it may vary based upon your needs.

the current software implementation of SPI gives some errors when using the interface simultaniously i am still somewhat working on that.

swerveshot, 2025-09-16:
If you want to utilize the bluetooth capabilities with something like the esphome bluetooth proxy, isn't flash size a concern? I would look more towards something like a [Adafruit ESP32 Feather V2](https://learn.adafruit.com/adafruit-esp32-feather-v2) which has 8 megs of flash.

ThomasHFWright, 2026-01-31:
> can someone with an esp32 and loop errors test this yaml and report how this works:
> 
> https://github.com/RobertJansen1/MHI-AC-Ctrl-ESPHome/blob/esp32_errors/examples/esp32_test.yaml

@RobertJansen1 Okay got some success!
Still seeing loop errors but now my unit is booting, can send commands successfully, and I even have the bluetooth proxy running.

I booted successfully the first time when setting ```board_build.f_cpu: 160000000L``` from ```240000000L```

I still got errors when setting the frame size to 20.

```LOG
INFO ESPHome 2026.1.3
INFO Reading configuration /config/esphome/airconoffice.yaml...
INFO Detected timezone 'Europe/Lisbon'
WARNING WiFi AP is configured but neither captive_portal nor web_server is enabled. The AP will not be usable for configuration or monitoring. Add 'captive_portal:' or 'web_server:' to your configuration.
INFO Starting log output from 192.168.1.183 using esphome API
INFO Successfully resolved airconoffice @ 192.168.1.183 in 0.000s
INFO Successfully connected to airconoffice @ 192.168.1.183 in 0.006s
INFO Successful handshake with airconoffice @ 192.168.1.183 in 0.174s
[16:41:45.901][I][app:206]: ESPHome version 2026.1.3 compiled on 2026-01-31 16:31:18 +0000
[16:41:45.901][I][app:213]: ESP32 Chip: ESP32 r3.1, 2 core(s)
[16:41:45.901][C][logger:316]: Logger:
[16:41:45.901][C][logger:316]:   Max Level: DEBUG
[16:41:45.901][C][logger:316]:   Initial Level: DEBUG
[16:41:45.913][C][logger:322]:   Log Baud Rate: 0
[16:41:45.913][C][logger:322]:   Hardware UART: UART0
[16:41:45.913][C][logger:332]:   Task Log Buffer Size: 768 bytes
[16:41:45.917][C][uptime.sensor:016]: Uptime Sensor 'Uptime'
[16:41:45.917][C][uptime.sensor:016]:   State Class: 'total_increasing'
[16:41:45.917][C][uptime.sensor:016]:   Unit of Measurement: 's'
[16:41:45.917][C][uptime.sensor:016]:   Accuracy Decimals: 0
[16:41:45.922][C][uptime.sensor:026]:   Device Class: 'duration'
[16:41:45.925][C][uptime.sensor:030]:   Icon: 'mdi:timer-outline'
[16:41:45.925][C][uptime.sensor:033]:   Type: Seconds
[16:41:46.015][C][bluetooth_proxy:124]: Bluetooth Proxy:
[16:41:46.015][C][bluetooth_proxy:124]:   Active: YES
[16:41:46.015][C][bluetooth_proxy:124]:   Connections: 3
[16:41:46.020][C][mhi.platform:069]: MHI Platform
[16:41:46.020][C][mhi.platform:071]:   external_temperature_sensor enabled!
[16:41:46.024][C][mhi.platform:074]:   frame_size: 33
[16:41:46.028][C][mhi.platform:075]:   room_temp_api_timeout: 60
[16:41:46.028][C][mhi.platform:076]:   listeners count: 8
[16:41:46.115][C][restart.button:014]: Restart Button 'Restart'
[16:41:46.120][C][restart.button:017]:   Icon: 'mdi:restart'
[16:41:46.120][C][mhi.climate:039]: MHI Climate
[16:41:46.150][C][homeassistant.time:010]: Home Assistant Time
[16:41:46.217][C][time:028]: Timezone: 'WET0WEST,M3.5.0/1,M10.5.0'
[16:41:46.217][C][mhi.binary_sensor:019]: MHI Binary Sensors
[16:41:46.217][C][mhi.binary_sensor:022]:   defrost: 0
[16:41:46.217][C][mhi.binary_sensor:025]:   vanes_3d_auto_enabled: 0
[16:41:46.217][C][mhi.sensor:037]: MHI Sensors
[16:41:46.217][C][mhi.sensor:039]:   error_code: 0.000000
[16:41:46.217][C][mhi.sensor:042]:   outdoor_temperature: 12.750000
[16:41:46.217][C][mhi.sensor:045]:   return_air_temperature: 21.250000
[16:41:46.217][C][mhi.sensor:048]:   outdoor_unit_fan_speed: 0.000000
[16:41:46.217][C][mhi.sensor:051]:   indoor_unit_fan_speed: 0.000000
[16:41:46.217][C][mhi.sensor:054]:   compressor_frequency: 0.000000
[16:41:46.218][C][mhi.sensor:057]:   indoor_unit_total_run_time: 1900.000000
[16:41:46.218][C][mhi.sensor:060]:   compressor_total_run_time: 3800.000000
[16:41:46.218][C][mhi.sensor:063]:   current_power: 0.000000
[16:41:46.220][C][mhi.sensor:066]:   vanes_pos: 5.000000
[16:41:46.220][C][mhi.sensor:069]:   energy_used: 0.000000
[16:41:46.220][C][mhi.sensor:072]:   indoor_unit_thi_r1: 20.319000
[16:41:46.227][C][mhi.sensor:075]:   indoor_unit_thi_r2: 19.000002
[16:41:46.227][C][mhi.sensor:078]:   indoor_unit_thi_r3: 20.646000
[16:41:46.234][C][mhi.sensor:081]:   outdoor_unit_tho_r1: 4.296000
[16:41:46.238][C][mhi.sensor:084]:   outdoor_unit_expansion_valve: 0.000000
[16:41:46.238][C][mhi.sensor:087]:   outdoor_unit_discharge_pipe: 38.000000
[16:41:46.238][C][mhi.sensor:090]:   outdoor_unit_discharge_pipe_super_heat: 7.000000
[... 300 lines cut ...]
  - platform: wifi_info
    ip_address:
      name: IP
    ssid:
      name: SSID
    bssid:
      name: BSSID
  - platform: MhiAcCtrl
    protection_state:
      name: "Compressor protection status"

select:
  - platform: MhiAcCtrl
    vertical_vanes:
      name: Fan Control Up Down
    horizontal_vanes:
      name: Fan Control Left Right
    fan_speed:
      name: "Fan Speed"

switch:
  - platform: MhiAcCtrl
    vanes_3d_auto:
      name: "3D Auto"
```