ginkage/MHI-AC-Ctrl-ESPHome #44: Current power - indoor and/or outdoor?

sanderlv, 2022-10-28:
Having a split unit and wondering if the current power is read from the outdoor unit and/or indoor unit(s).

I have 2 indoor units and thus 2 esps and reading different values from both (might have to do with the refresh rate not being synchronous).

Thanks.

synegic, 2022-10-30:
AFAIK current is always measured at the outdoor unit. I don't think it is possible to split the currents between the multiple units in a good way. Maybe if one indoor unit is turned off, the other will report a value but the one being off will not. In this case the values are of course different. But if they are both running, they should in theory display the same outdoor unit values.

But then again; no idea. I don't have a split unit at home. Again; this question touches on the internals of the AC and the data exchanged, so the better place to ask this is on the core library's git.

ervee, 2022-10-30:
Yes, the current value is from the outside unit and should report about the same on both indoor units. Just as 5 other entities (compressor frequency/runtime, defrost, outdoor temp/fan). I just disabled these 6 entities in HA. You could even strip these sensors on all but one unit of a split system but I didn't bother.

sanderlv, 2022-10-30:
Thanks for confirming

closed 2022-10-30 by sanderlv