ginkage/MHI-AC-Ctrl-ESPHome #46: Vane up/down control resets vane left/right

ervee, 2022-11-16:
Hi, in #38 vane control was added but I have a weird issue with this and I wonder if I'm the only one :)

When I set my left/right vane (with the IR) and then later call the new vane control service to set my up/down vane to something, let's say "4", the vane goes to the bottom as expected but the left/right vanes reset to the middle!

Could it perhaps be that the signal sent for the new up/down vane control also includes a left/right position?

Also tagging https://github.com/absalom-muc/MHI-AC-Ctrl/issues/77 here because there is some development going on there.

JoepWi, 2023-06-23:
I got the same.
Would be nice if we also could set the l/r movement and maybe also the 3D setting.

JoepWi, 2023-07-14:
Is there any progress on the left/right challenge?

tismij, 2023-07-14:
how do i add the option to set the vanes up/down ?
Updated to latest version but in HA vanes show as a sensor and do not seems to grasp how to add a control to change this.

RobertJansen1, 2023-10-30:
this was fixed in https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/pull/63 the code for left/right vanes was wrong as well so the behaviour you explained was kinda how it stood in code.

closed 2024-01-18 by RobertJansen1