ginkage/MHI-AC-Ctrl-ESPHome #47: Control fanspeed and swingposition through webserver

DYLaKo, 2022-11-30:
In my ESPHome code, I have enabled the webserver for my module:

```
web_server:
  port: 80
```

In the webserver, all sensor data available in HomeAssistant is also shown here. The controls for the AC are very limited though.
It is only possible to set the temperature and the mode. Within HomeAssistant I can also change the fanspeed and swingposition, which are attributes of the climate component.

Is it possible to make fanspeed and swingposition also available in the webserver?

ddrag001, 2022-12-05:
I am also interested in this. Why not implement it if it is possible?

synegic, 2022-12-05:
The controls seem to be missing in the default webserver Climate device JS code. Not much you can do about that. In theory you can hack around it by creating some extra templates/virtual components to execute the callbacks manually.

For example, using the select component (https://esphome.io/components/select/index.html);

```
web_server:
  port: 80
  include_internal: true

select:
  - platform: template
    id: vanes_select
    name: "Vanes Position Override"
    optimistic: true
    internal: true
    options:
      - unknown
      - one
      - two
      - three
      - four
      - swing
    initial_option: unknown
    on_value:
      then:
        - lambda: ((MhiAcCtrl*)id(${deviceid}))->set_vanes((i-1));
  - platform: template
    id: fan_speed
    name: "Fan Speed Override"
    optimistic: true
    internal: true
    options:
      - low
      - medium
      - high
      - auto
    initial_option: auto
    on_value:
      then:
        - climate.control:
            id: ${deviceid}
            fan_mode: !lambda |-
                if (i == 0) {
                  return CLIMATE_FAN_LOW;
                } else if (i == 1) {
                  return CLIMATE_FAN_MEDIUM;
                } else if (i == 2) {
                  return CLIMATE_FAN_HIGH;
                } else {
                  return CLIMATE_FAN_AUTO;
                }
```

This does not support updating the value in the list to the actual value, so it is really only an override option. Adding such functionality would require more code and maybe even some changes inside the component code.

WorlockM, 2022-12-30:
> 

I tried this code but I don't see any changes in the web interface?

synegic, 2022-12-30:
> I tried this code but I don't see any changes in the web interface?
You need to set `include_internal: true` in your webserver section else the template items will not show up in the web interface.

WorlockM, 2022-12-30:
> include_internal: true

Ah yes I overlooked that one, thanks!

closed 2024-04-03 by RobertJansen1