ginkage/MHI-AC-Ctrl-ESPHome #50: DS18B20 as external temperature sensor

DYLaKo, 2022-12-20:
The original MHI-AC-Ctrl software and PCB have an option to connect a DS18B20 temperature sensor and use that sensor instead of the internal temperature sensor of the HVAC unit.
(upper left corner of: https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/images/PCB_top.PNG)
(https://github.com/absalom-muc/MHI-AC-Ctrl/blob/master/SW-Configuration.md#external-temperature-sensor-settings-supporth)

In the MHI-AC-Ctrl-ESPHome version, I couldn't find the option to use a connected DS18B20 temperature sensor. 

Is it possible to add this feature to the code?

glsf91, 2022-12-21:
You can add this in the yaml file:

```
# Dallas 18x20 temperature sensor configuration entry
dallas:
  - pin: GPIO4    # D2 on a Wemos D1 mini (8266)
    update_interval: 60s

  - platform: dallas
   # index accept any found single Dallas sensor.
   # Use address, when you have more than one Dallas sensors hooked up at the same time
    index: 0
   # address: 0x0123456789ABCDEF
    state_class: measurement # keep statisics about the sensor in Home Assistent. Not available in stand stand alone (ESPhome alone)
    id: troom_dallas
   # on_value enables to use then: and "lambda: |-" to able to write C++ code in the YAML source
   # this is the overwrite of the internal sensor with the one connected directly to you intervacel MHI-AC-CTL board and Wemos D1 Mini shield 
    on_value:
      then:
          - lambda: |-
              float value;
              value = id(troom_dallas).state;
              ESP_LOGD("mhi_ac_ctrl", " Room temperature sensor value: %f °C", value);
              return ((MhiAcCtrl*)id(${deviceid}))->set_room_temperature(value);

  # If you still want to have the value being published to the outside
  - platform: template
    name: ${devicename} Room Temperature
    id: "room_temperature"
    unit_of_measurement: "°C"
    lambda: return id(troom_dallas).state;
    
```

(copied from someone else)

DYLaKo, 2022-12-21:
Amazing! Thank you.

closed 2024-08-30 by RobertJansen1