ginkage/MHI-AC-Ctrl-ESPHome #51: MHI-AC-Ctrl v2.6 update?

JoepWi, 2023-01-29:
Can we incorporate all updates from this version?: https://github.com/absalom-muc/MHI-AC-Ctrl/releases/tag/v2.6

Kurisutian, 2023-02-07:
Any information when there will be an upgrade? Especially the changes around the external thermostat and the fan auto mode are great improvements. 😃

Bill-Dung, 2023-02-28:
The main problem will be that ESPHome only supports three fan speeds + auto, but the MHI ACs have four speeds and auto.

Kurisutian, 2023-03-01:
So basically you're saing that it won't be possible to upgrade to 2.6 unless this shortcoming has been addressed to and fixed by the ESPHome team? Just wondering as it seems that someone did just used the code, created a component to integrate in ESPHome and upgraded to 2.6 (https://github.com/tfyoung/esphome-MHIHeatPump). 
However, checking it out and using that code did not work for me.

Bill-Dung, 2023-03-03:
I didn't say it's impossible, you have to create a component and extend the climate class, this cannot be done on passing by. The mentioned component didn't work for me either, it seems to be cobbled together quickly.

ervee, 2023-03-03:
What about this solution: https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/issues/22#issuecomment-1310271934

Bill-Dung, 2023-03-04:
Sounds interesting, I'll give a try next week. If it works I could create a fork of MHI-AC-Ctrl-ESPHome so everyone can use it until ginkage makes an update available.

Bill-Dung, 2023-03-06:
Find it here: https://github.com/Bill-Dung/MHI-AC-Ctrl-ESPHome
AC has now 5 different fan modes but I'm not sure if the auto mode works proper, keep testing.

Kurisutian, 2023-03-06:
> Find it here: https://github.com/Bill-Dung/MHI-AC-Ctrl-ESPHome AC has now 5 different fan modes but I'm not sure if the auto mode works proper, keep testing.

Thanks, I will test it. Is the header file the only thing which changed compared to this repo? So it would be v2.5, correct?

Bill-Dung, 2023-03-06:
Yes, it's 2.5

ervee, 2023-03-06:
I will also look into this now I see how this change is done. Thank you. And if it works I or Bill-Dung or someone could make a pull request. Perhaps using the CLIMATE_FAN_QUIET instead of DIFFUSE. CLIMATE_FAN_QUIET should also be available in climate and esphome and is perhaps a more logical choice for ultra-low fan speed.

Bill-Dung, 2023-03-07:
CLIMATE_FAN_QUIET is not be showing up in Home Assistant Card like CLIMATE_FAN_MIDDLE and CLIMATE_FAN_FOCUS. If we want to go more into the deep of this please open an issue on my fork.

RobertJansen1, 2023-06-22:
CLIMATE_FAN_QUIET is now implemented in home assistant: https://github.com/home-assistant/core/commit/d14000f7049f955a08b9c164798f5f78335e54b6

i created a pull request implementing both v2.7R4 and the fan modes from @Bill-Dung, adjusting the DIFFUSE to Quiet. 

https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/pull/54

testing can be done from https://github.com/RobertJansen1/MHI-AC-Ctrl-ESPHome

closed 2023-07-14 by JoepWi

serrj-sv, 2023-08-04:
> CLIMATE_FAN_QUIET is now implemented in home assistant: [home-assistant/core@d14000f](https://github.com/home-assistant/core/commit/d14000f7049f955a08b9c164798f5f78335e54b6)
> 
> i created a pull request implementing both v2.7R4 and the fan modes from @Bill-Dung, adjusting the DIFFUSE to Quiet.
> 
> #54
> 
> testing can be done from https://github.com/RobertJansen1/MHI-AC-Ctrl-ESPHome

I don't like the fact that "quiet" comes after "high" in Dashboard card. I propose to change this a bit, see: 
https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/pull/62