ginkage/MHI-AC-Ctrl-ESPHome #52: Since ESPheom update I get an error: warning: 'esphome::climate::Climate::away' is deprecated: away is deprecated, use preset instead [-Wdeprecated-declarations]

sanderlv, 2023-04-29:
Compiling /data/aircozolder/.pioenvs/aircozolder/libaf0/Hash/Hash.cpp.o
In file included from src/esphome/core/controller.h:29,
                 from src/esphome/components/api/api_server.h:4,
                 from src/esphome/components/api/api_connection.h:6,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/components/climate/climate.h: In constructor 'esphome::climate::Climate::Climate()':
src/esphome/components/climate/climate.h:167:7: warning: 'esphome::climate::Climate::away' is deprecated: away is deprecated, use preset instead [-Wdeprecated-declarations]
  167 | class Climate : public EntityBase {
      |       ^~~~~~~
src/esphome/components/climate/climate.h:193:8: note: declared here
  193 |   bool away{false};
      |        ^~~~
src/esphome/components/climate/climate.h:167:7: warning: 'esphome::climate::Climate::away' is deprecated: away is deprecated, use preset instead [-Wdeprecated-declarations]
  167 | class Climate : public EntityBase {
      |       ^~~~~~~
src/esphome/components/climate/climate.h:193:8: note: declared here
  193 |   bool away{false};
      |        ^~~~
In file included from src/main.cpp:64:
src/mhi_ac_ctrl.h: In constructor 'MhiAcCtrl::MhiAcCtrl()':
src/mhi_ac_ctrl.h:8:7: note: synthesized method 'esphome::climate::Climate::Climate()' first required here
    8 | class MhiAcCtrl : public climate::Climate,
      |       ^~~~~~~~~
/config/esphome/aircozolder.yaml: In lambda function:
/config/esphome/aircozolder.yaml:65:40: note: synthesized method 'MhiAcCtrl::MhiAcCtrl()' first required here
   65 |       auto mhi_ac_ctrl = new MhiAcCtrl();
      |                                        ^
Compiling /data/aircozolder/.pioenvs/aircozolder/lib67b/ESP8266WiFi/BearSSLHelpers.cpp.o

Theetjuh, 2023-05-01:
Yeah getting the same message since the newer 2023 versions of ESPHome.

sanderlv, 2023-06-10:
Any progress possible?

ervee, 2023-06-10:
Does it still compile and run correctly? I've been holding off this update since I saw this issue logged.

RobertJansen1, 2024-03-30:
I don't have these errors, is this still an issue?

closed 2024-04-03 by RobertJansen1