ginkage/MHI-AC-Ctrl-ESPHome #57: Can't set airco mode in web server

WorlockM, 2023-06-22:
Since today on all my devices I am not able to select the airco mode (Cooling, heating, fan only etc in the web server anymore.
Even if I upload a old .bin file this functionality is gone.

WorlockM, 2023-06-23:
Someone else I know has the exact same thing. The devices are on ESPHome 2023.4 version so it doesn't seem like it has updated or something, but it all started at the same time. The option is just gone. Instead it now says state "24".

JoepWi, 2023-06-23:
I just checked and indeed the webserver has no possibility anymore to do settings etc.

WorlockM, 2023-06-25:
It probabely has to do with [this](https://github.com/esphome/esphome-webserver/pull/29/commits/af61e52ff31d32d1d21a8d258eb495e09bf1c80d#diff-27af2b81eb9dfc2444092b9a50036bf8b127d3a57f3913bbc1f741f95b0d3955R61) change on ESPHome, so I think the issue can't be solved by this Repo and therefore may be closed. Unless there's a workarround that can be provided?

closed 2023-09-08 by WorlockM