ginkage/MHI-AC-Ctrl-ESPHome #59: Home Assistant 2023.7 requires a setting to make service calls from ESPHome

ervee, 2023-07-09:
With pull https://github.com/home-assistant/core/pull/95143 for Home Assistant 2023.7, they now require a setting per ESPHome device to allow service calls.

In lr_mhi_ac_ctrl.yaml you use a call to set_api_room_temperature.
So in case anyone uses this for new devices starting Home Assistant 2023.7 you need to enable the ESP device to make the service call.
[image]

Perhaps update the yaml example and/or the main README with a note?

arpiecodes, 2023-07-09:
As far as I understand this new feature, it's for calling services from devices, not from HA. The `set_api_room_temperature` is actually a service exposed by the device, not HA. So this option does not have to be enabled for the 'use case' to update room temp from HA.

JoepWi, 2023-07-09:
I can confirm that set_api_room_temperature just works fine.
As @arpiecodes states. The new functionality had to do with info going back from devices as an answer to specific HA call.

ervee, 2023-07-09:
Great. Thanks for the confirmation!

ervee, 2023-07-09:
Close

ervee, 2023-07-09:
Woops

closed 2023-07-09 by ervee