ginkage/MHI-AC-Ctrl-ESPHome #60: MHI-AC-Ctrl working MQTT communication with SRK 63 ZR, but MHI-AC-Ctrl-ESPHome does not show any communication

the-master006, 2023-07-10:
I tried first to run my SRK 63 ZR with the D1 Wemos with MHI-AC-Ctrl-ESPHome, this does nothing besides connecting with Wifi. I flashed this via the python interface from ESPHome. 
Afterwards I tried to flash the D1 Wemos with the original MHI-AC-Ctrl project and then I can clearly see the MQTT communication working in the MQTT explorer.

What is going wrong? I saw somewhere in the files that it includes arduino.h, but I did not put that file in the directory, might that be the issue? Or anything else?
See attached the MQTT communication from the MHI-AC-Ctrl program.
Thanks!
[image]

the-master006, 2023-07-26:
Any feedback?

RobertJansen1, 2023-10-03:
what is the content of your esphome yaml file? did you include:
esphome:
  includes:
    - mhi_ac_ctrl.h
    - MHI-AC-Ctrl-core.h
    - MHI-AC-Ctrl-core.cpp

the-master006, 2023-10-04:
Ok, thanks for your answer. I will check!

closed 2024-01-25 by RobertJansen1