ginkage/MHI-AC-Ctrl-ESPHome #65: It is getting much warmer then setpoint

ervee, 2023-10-22:
Hi! I'm not sure whether this is a bug yet but I'd love to hear your experience and perhaps we can figure this out.

I don't use the HA integration during cooling season a lot, the previous heating season was a while ago and I updated to the latest version with a lot of changes so my past experience isn't that useful.

So I just started using the heating of my AC again and notice it is getting much warmer then I expect.

[image]

I have the target temperature set on 19 C. After a while the temperature in HA on the climate device shows 22 or 22.5 C.

[image]

I have a couple of devices in the same area reporting temperature at 20 to 20.5 C.

I don't mind that the climate device in HA shows too high of a temperature, but the room is actually getting too warm.

I need to test with IR remote only to see if this is also the case when I set the temperature with the remote. But how is your experience with this?

JoepWi, 2023-10-22:
> Hi! I'm not sure whether this is a bug yet but I'd love to hear your experience and perhaps we can figure this out.
>
> I don't use the HA integration during cooling season a lot, the previous heating [...]

For this exact reason I implemented external temp-sensors and I let HA turn off the AC-unit when the temp exceeds a certain threshold. I do the same btw with cooling.

SomeSander, 2023-10-22:
Did you again check the setpoint as with the offset it jumps 2 degrees as default. It can be disabled

[image]

arpiecodes, 2023-10-22:
I also noticed if the unit it set to heat (or cool for that matter) it overshoots its target. I typically solve it by lowering the setpoint once a base temp has been reached by 2 degrees in an automation to keep it constant from there on out. I think the unit compensates for the fact the temp sensor is in the unit itself, even if you are using an external sensor. Which makes sense, since warm air rises.

I also noticed when using an external temp sensor in auto mode, results are much better. It will then also stop operation if setpoint has been reached. But your mileage may vary.

hvorragend, 2023-10-22:
Now things get very interesting. I have two identical indoor units. One with built-in Wifi module and one with the ESP solution. I also have the feeling that the device with the ESP doesn't let out as much cold air.

ervee, 2023-10-22:
> For this exact reason I implemented external temp-sensors and I let HA turn off the AC-unit when the temp exceeds a certain threshold. I do the same btw with cooling.

I tried this last year (with older code off course) and that had some other issues for me but I don't remember what exactly. I could give this another try. I have 3 other sensors in a min/max helper that I could use. I would never trust a single sensor with this task.

It was also possible to offset the temperature somewhere but this resulted in incorrect temperature reporting to the climate integration or something.

SomeSander, 2023-10-22:
> > Did you check the setpoint as with the offset it jumps 2 degrees as default. It can be disabled
> > [image]
> 
> Do you press the button on the indoor unit or on the IR remote? The manual seems a bit unclear about that.

On the unit itself

ervee, 2023-10-22:
> On the unit itself

Okay. When I do this when the unit is off it just goes on and off over and over. So I tried when the unit is heating and then after pressing the button on the underside for 30 second it quickly beeped 2 times 😂 Not one pip or 3 but I'll see what this does.

SomeSander, 2023-10-22:
The information is a bit off indeed. But it helped to keep my setpoint stable afterwards. I had an identical journey to see why my temperature kept raising by 2 degrees. I only don't use the cool_heat combination myself, as this might switch too often without this setting

ervee, 2023-11-22:
I'm using my AC for a month now with this temperature correction and it is much better. It is now undershooting the set temperature just a bit bit I like it better this way.

I have 3 ZigBee temperature sensors it my room which give me a nice average temperature with a mix/max helper so I'm now going to feed this temperature back to the AC ESP device and test how this affects the AC temperature.

closed 2024-02-01 by RobertJansen1

ervee, 2024-02-06:
I do have to comment after closing that feeding the external temperature data to the AC make the thing go completely off the rails and keep heating until I'm in my underwear. So I stopped doing that. I'll perhaps try this again later and if it still does that create another issue here.