ginkage/MHI-AC-Ctrl-ESPHome #66: Using a Honeywell T6 for the room temperature

MarceldeJager, 2023-10-24:
Hi There, 

I am bit new in this all and when I asked a stupid question I'm sorry :-)

I have two AC's.  One in the livingroom and one in the bedroom. Both have the wifi controller with MHI-AC-Ctrl-ESPHome software on it and they work fine (after 2 weeks of messing around).
Setting the temp is aways a guess because depending on heating, or cooling, the offset is different.
In the living room I have the Honeywell T6 that is attached the the normal heater (is hardly used)  and that always has the right temp.
Is it possible to use the Honeywel as the room temp by the airco and how do I do that? 

Thanks,

Marcel

ervee, 2023-10-24:
Do you already have the Honeywell as a temperature sensor in Home Assistant?
If so you can feed this sensor into the AC as external temperature via the service set_api_room_temperature. You need to call this regularly within the timeout specified in the ESPHome yaml for your AC.

arpiecodes, 2024-10-24:
> One thing I noticed is that the temperature reading is interpreted as being 2 degrees under target. So when I set the AC to 20 degrees, it looks like it is aiming for a reported room temperature of about 22 degrees. I do notice that the AC is running with less pauses since I switched to using a room thermometer instead of the built in one.

This is expected behaviour. See https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/issues/65#issuecomment-1774071728 on how to change this. It's in the manual. If you follow those steps mentioned there it will no longer overshoot.

RobertJansen1, 2025-07-16:
this was fixed by the awesome work of @LawnMeower in #133 

see [`external_sensor.yaml`](https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/blob/master/examples/external_sensor.yaml) for an example on how to do this