ginkage/MHI-AC-Ctrl-ESPHome #67: Bluetooth temp sensor

eddydordt, 2023-10-24:
Hello,

I have updated from the MQTT version to esphome. In MQTT, I could write the temperature from a Bluetooth sensor to the unit.
When I want to do this now, I get an error:

This service requires field value, please enter a valid value for value.

[image]

Can someone tell me what's going wrong?

JoepWi, 2023-10-24:
Add "| float" before the "}}"

eddydordt, 2023-10-25:
Like this:

[image]

This still doesn't work

JoepWi, 2023-10-25:
> Like this:
> 
> 
> 
> [image]
> 
> 
> 
> This still doesn't work
> 
> 

Change to yaml mode and then do it like this:
services: esphome.mhi_YOURDEVICENAME_api_room_temperature
data:
  value: "{{ states('YOURDEVICENAME') | float }}"

eddydordt, 2023-10-25:
I still get an error:

[image]

eddydordt, 2023-10-25:
I found the solution, the indent was the problem:

[image]

Thx for the help

persistive, 2024-06-02:
Edddydort,

I have copied your solution to use a external wifi temp reading as TRoom value.
The problem is that i need to hit the "call service" button to update the external temperature reading.
Am i doing something wrong?
[image]

eddydordt, 2024-06-02:
You need to put this service in automation. And let it publish every 30 seconds or so

persistive, 2024-06-02:
Thanks Eddydordt,

I've managed to get it to work allthough it feels like a "way arround".
Would it not be better to directly lead the Troom input to an available temperature entity in HA?
Nevertheless many thanks for your help.

Dennis-Q, 2024-10-04:
I think ESPHome has the ability to read sensors directly from HA and use them, but then you would need to create the automation in ESPHome. In reality, I don't think it's going to be more reliable since you still need your home assistant to be available.

The only more neat solution would be a wired temperature sensor or a bluetooth sensor which is directly connected to the ESP that controls your AC.. But bluetooth/wifi share the same antenna on ESPHome so it could influence the stability of the connectivity.. And then you still need an automation within ESPHome as there's no native implementation currently.