ginkage/MHI-AC-Ctrl-ESPHome #68: AC working with MHI-AC-Ctrl but not with MHI-AC-Ctrl-ESPHome

Stimuka, 2023-11-10:
Hello,

Two days ago I installed my hardware in the AC unit which is a SRK20ZM-S and I uploaded the MHI-AC-Ctrl-ESPHome software on the ESP8266 which is a D1 Mini. Unfortunately I couldn't get it to work. All my sensors gave unknown values and also I didn't get the current temperature value or couldn't change HVAC modes.

Below is the output of the console when the MHI-AC-Ctrl-ESPHome software is active.

[I][wifi:519]: WiFi Connected!
[C][wifi:363]:   Local MAC: F4:CF:A2:65:FA:CD
[C][wifi:364]:   SSID: '*******'
[C][wifi:365]:   IP Address: *******
[C][wifi:366]:   BSSID: *******
[C][wifi:368]:   Hostname: 'lr_mhi_ac_ctrl'
[C][wifi:370]:   Signal strength: -76 dB
[C][wifi:374]:   Channel: 6
[C][wifi:375]:   Subnet: 255.255.255.0
[C][wifi:376]:   Gateway: 192.168.5.1
[C][wifi:377]:   DNS1: 192.168.1.5
[C][wifi:378]:   DNS2: 192.168.1.6
[D][wifi:528]: Disabling AP...
[W][mhi_ac_ctrl:139]: mhi_ac_ctrl_core.loop error: -2
[C][ota:093]: Over-The-Air Updates:
[C][ota:094]:   Address: lr_mhi_ac_ctrl.local:8266
[C][api:025]: Setting up Home Assistant API server...
[I][app:062]: setup() finished successfully!
[D][sensor:109]: 'Airco Slaapkamer WiFi Signal': Sending state -76.00000 dBm with 0 decimals of accuracy
[D][text_sensor:064]: 'Airco Slaapkamer BSSID': Sending state '06:EC:DA:3B:F8:E1'
[D][text_sensor:064]: 'Airco Slaapkamer SSID': Sending state '*******'
[D][text_sensor:064]: 'Airco Slaapkamer IP': Sending state '192.168.5.54'
[W][mhi_ac_ctrl:139]: mhi_ac_ctrl_core.loop error: -2
[I][app:102]: ESPHome version 2023.4.4 compiled on Nov 10 2023, 09:27:45
[C][wifi:505]: WiFi:
[C][wifi:363]:   Local MAC: F4:CF:A2:65:FA:CD
[C][wifi:364]:   SSID: '*******'
[C][wifi:365]:   IP Address: 192.168.5.54
[C][wifi:366]:   BSSID: 5m06:EC:DA:3B:F8:E16m
[C][wifi:368]:   Hostname: 'lr_mhi_ac_ctrl'
[C][wifi:370]:   Signal strength: -76 dB
[C][wifi:374]:   Channel: 6
[C][wifi:375]:   Subnet: 255.255.255.0
[C][wifi:376]:   Gateway: 192.168.5.1
[C][wifi:377]:   DNS1: 192.168.1.5
[C][wifi:378]:   DNS2: 192.168.1.6
[W][mhi_ac_ctrl:139]: mhi_ac_ctrl_core.loop error: -2
[C][logger:294]: Logger:
[C][logger:295]:   Level: DEBUG
[C][logger:296]:   Log Baud Rate: 115200
[C][logger:297]:   Hardware UART: UART0
[W][mhi_ac_ctrl:139]: mhi_ac_ctrl_core.loop error: -2
[W][mhi_ac_ctrl:139]: mhi_ac_ctrl_core.loop error: -2
[C][uptime.sensor:031]: Uptime Sensor 'Airco Slaapkamer Uptime'
[C][uptime.sensor:031]:   Device Class: 'duration'
[C][uptime.sensor:031]:   State Class: 'total_increasing'
[C][uptime.sensor:031]:   Unit of Measurement: 's'
[C][uptime.sensor:031]:   Accuracy Decimals: 0
[C][uptime.sensor:031]:   Icon: 'mdi:timer-outline'
[W][mhi_ac_ctrl:139]: mhi_ac_ctrl_core.loop error: -2
[C][template.select:065]: Template Select 'Airco Slaapkamer Fan Control Left Right'
[C][template.select:065]:   Icon: 'mdi:arrow-left-right'
[C][template.select:066]:   Update Interval: 60.0s
[C][template.select:069]:   Optimistic: YES
[C][template.select:070]:   Initial Option: Left
[C][template.select:071]:   Restore Value: NO
[W][mhi_ac_ctrl:139]: mhi_ac_ctrl_core.loop error: -2
[C][template.select:065]: Template Select 'Airco Slaapkamer Fan Control Up Down'
[C][template.select:065]:   Icon: 'mdi:arrow-up-down'
[C][template.select:066]:   Update Interval: 60.0s
[C][template.select:069]:   Optimistic: YES
[C][template.select:070]:   Initial Option: Up
[C][template.select:071]:   Restore Value: NO
[W][mhi_ac_ctrl:139]: mhi_ac_ctrl_core.loop error: -2
[... 135 lines cut ...]
mhi_ac_ctrl_core.loop error: -4
status=132 topic=OUTDOOR payload=6.25
mhi_ac_ctrl_core.loop error: -2
mhi_ac_ctrl_core.loop error: -2
status=133 topic=THO-R1 payload=0
mhi_ac_ctrl_core.loop error: -2
status=141 topic=COMP payload=0.00
mhi_ac_ctrl_core.loop error: -2
status=143 topic=TD payload=<=30
mhi_ac_ctrl_core.loop error: -1
status=142 topic=CT payload=0.00
mhi_ac_ctrl_core.loop error: -4
mhi_ac_ctrl_core.loop error: -4
mhi_ac_ctrl_core.loop error: -2
status=144 topic=TDSH payload=0
status=145 topic=PROTECTION-NO payload=0
status=138 topic=OU-FANSPEED payload=0
mhi_ac_ctrl_core.loop error: -2
status=146 topic=DEFROST payload=Off
status=140 topic=TOTAL-COMP-RUN payload=1800
mhi_ac_ctrl_core.loop error: -4

What I have done so far is use different ESPHome versions. The versions I have used so far are 2023.2.0, 2023.4.4 and the latest. I have also used the MHI-AC-Ctrl-core.cpp and MHI-AC-Ctrl-core.h from the MHI-AC-Ctrl which was working for me. But nothing did the trick thus far. 

One strange thing I saw from Home Assistant is that after about an hour of being connected some sensor are are filled. The current temperature seems to be 19 degrees and does not change over a full night. Also the error code sensor got a value of 1.

damienmorrissey, 2023-11-13:
I had the same problem. I didn't persist as long as you did (because I had the benefit of your post), however once I flashed the MHI-AC-Ctrl firmware it worked right away. I keep everything up to date so it was the latest version of esphome.

Something I was wondering- could the debug messages that have been added regarding timing be causing timing issues?

I have five units. I will try to get the model numbers. The three most modern units I have had a chance to install the modules on- the other two need to wait a couple of days as it's inconvenient to get to them. Out of the five I have four different models!

Stimuka, 2023-11-13:
I am not sure if that is the problem. Last night however I installed the fork from @arpiecodes and everything seems to run fine. So there must be something significantly changed in one of the commits in this repository I think.

AlexzeyK, 2023-11-15:
Guys, I think I found a solution to the problem.
take a look   @https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/issues/69

damienmorrissey, 2024-01-10:
> This is because of the addition of 3dvanes which is not working on older
> units (and the bigger framesize is now the default)
>
> if you install
> https://github.com/RobertJansen1/MHI-AC-Ctrl-ESPHome/tree/legacy_support
> and change the framesize to 20, the older units will continue working.
>
> i patched the files and added an option to configure this setting from the
> yaml file. if this can be tested, it would be greatly appreciated and i can
> add documentation for it if this works :)

Team,
I can confirm that this is indeed the case. All of my units (I have 5) need
the smaller frame size.

I guess we could be smart and adjust the size of it fails. Best of both
worlds? I might try to add that...

Dennis-Q, 2024-01-10:
@damienmorrissey it would be great if you can, but for now I am happy it's configurable via the esphome yaml.

In #69 I shared my experience regarding old and new units and the frame size and confirmed the legacy fix provided by @RobertJansen1 to be working.

Would you be able to share the model of your indoor units in the other issue mentioned so we het a list of legacy models not supporting the new frame size?

I do think it's good to  consolidate both issues into one ticket as they seem to be duplicates.