ginkage/MHI-AC-Ctrl-ESPHome #69: [mhi_ac_ctrl:139]: mhi_ac_ctrl_core.loop error: -2: Problem and solution

AlexzeyK, 2023-11-15:
First of all, I want to thank the author for the excellent adaptation of this solution in ESPHome.
version 2.8 added much-needed components. This is the control of vanes).
However, it added a problem for my device  SRK 63 ZSPR-S.
I upgraded from 2.7 to 2.8 and got the error:  [mhi_ac_ctrl:139]: mhi_ac_ctrl_core.loop error: -2
this error made the integration not working.

`INFO ESPHome 2023.11.0
INFO Reading configuration /config/esphome/ac-ctrl-mits2.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing ac-ctrl--mits2 (board: d1_mini; framework: arduino; platform: platformio/espressif8266@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 160MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/ac-ctrl--mits2/src/main.cpp.o
Linking .pioenvs/ac-ctrl--mits2/firmware.elf
RAM:   [=====     ]  46.0% (used 37676 bytes from 81920 bytes)
Flash: [=====     ]  47.7% (used 498401 bytes from 1044464 bytes)
Building .pioenvs/ac-ctrl--mits2/firmware.bin
esp8266_copy_factory_bin([".pioenvs/ac-ctrl--mits2/firmware.bin"], [".pioenvs/ac-ctrl--mits2/firmware.elf"])
========================= [SUCCESS] Took 13.13 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of ac-ctrl--mits2.local
INFO  -> 192.168.21.7
INFO Uploading /data/build/ac-ctrl--mits2/.pioenvs/ac-ctrl--mits2/firmware.bin (502560 bytes)
INFO Compressed to 341194 bytes
Uploading: [============================================================] 100% Done...
INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from ac-ctrl--mits2.local using esphome API
WARNING Can't connect to ESPHome API for ac-ctrl--mits2: Error resolving IP address: [Errno -5] No address associated with hostname (APIConnectionError)
INFO Trying to connect to ac-ctrl--mits2 in the background
INFO Successfully connected to ac-ctrl--mits2 in 0.013s
INFO Successful handshake with ac-ctrl--mits2 in 0.235s
[11:34:26][I][app:102]: ESPHome version 2023.11.0 compiled on Nov 15 2023, 11:33:50
[11:34:26][C][wifi:559]: WiFi:
[11:34:26][C][wifi:391]:   Local MAC: C8:C9:A3:54:89:5E
[11:34:26][C][wifi:396]:   SSID: [redacted]
[11:34:26][C][wifi:397]:   IP Address: 192.168.21.7
[11:34:26][C][wifi:398]:   BSSID: [redacted]
[11:34:26][C][wifi:400]:   Hostname: 'ac-ctrl--mits2'
[11:34:26][C][wifi:402]:   Signal strength: -41 dB ▂▄▆█
[11:34:26][C][wifi:406]:   Channel: 11
[11:34:26][C][wifi:407]:   Subnet: 255.255.255.0
[11:34:26][C][wifi:408]:   Gateway: 192.168.21.1
[11:34:26][C][wifi:409]:   DNS1: 192.168.21.1
[11:34:26][C][wifi:410]:   DNS2: 93.175.192.21
[11:34:26][W][mhi_ac_ctrl:139]: mhi_ac_ctrl_core.loop error: -2
[11:34:26][W][component:214]: Component <unknown> took a long time for an operation (0.06 s).
[11:34:26][W][component:215]: Components should block for at most 20-30ms.
[11:34:26][C][logger:416]: Logger:
[11:34:26][C][logger:417]:   Level: DEBUG
[11:34:26][C][logger:418]:   Log Baud Rate: 0
[11:34:26][C][logger:420]:   Hardware UART: UART0
[11:34:26][W][mhi_ac_ctrl:139]: mhi_ac_ctrl_core.loop error: -2
[11:34:26][W][component:214]: Component <unknown> took a long time for an operation (0.06 s).
[11:34:26][W][component:215]: Components should block for at most 20-30ms.
[11:34:27][W][mhi_ac_ctrl:139]: mhi_ac_ctrl_core.loop error: -2
[11:34:27][W][component:214]: Component <unknown> took a long time for an operation (0.07 s).
[11:34:27][W][component:215]: Components should block for at most 20-30ms.
[11:34:27][C][uptime.sensor:031]: Uptime Sensor 'MHI-AC-Ctrl__2 Uptime'
[11:34:27][C][uptime.sensor:031]:   Device Class: 'duration'
[11:34:27][C][uptime.sensor:031]:   State Class: 'total_increasing'
[... 290 lines cut ...]
[11:45:21][D][climate:400]:   Swing Mode: OFF
[11:45:21][D][climate:403]:   Current Temperature: 21.50°C
[11:45:21][D][climate:409]:   Target Temperature: 18.00°C
[11:45:22][D][mhi_ac_ctrl:154]: received status=144 value=0 power=0
[11:45:22][D][climate:380]: 'MHI-AC-Ctrl__2' - Sending state:
[11:45:22][D][climate:383]:   Mode: OFF
[11:45:22][D][climate:388]:   Fan Mode: HIGH
[11:45:22][D][climate:400]:   Swing Mode: OFF
[11:45:22][D][climate:403]:   Current Temperature: 21.50°C
[11:45:22][D][climate:409]:   Target Temperature: 18.00°C
[11:45:22][D][sensor:093]: 'MHI-AC-Ctrl__2 Outdoor unit discharge pipe super heat': Sending state 0.00000 °C with 1 decimals of accuracy
[11:45:23][D][mhi_ac_ctrl:154]: received status=145 value=0 power=0
[11:45:23][D][climate:380]: 'MHI-AC-Ctrl__2' - Sending state:
[11:45:23][D][climate:383]:   Mode: OFF
[11:45:23][D][climate:388]:   Fan Mode: HIGH
[11:45:23][D][climate:400]:   Swing Mode: OFF
[11:45:23][D][climate:403]:   Current Temperature: 21.50°C
[11:45:23][D][climate:409]:   Target Temperature: 18.00°C
[11:45:23][D][text_sensor:064]: 'MHI-AC-Ctrl__2 Protection error state': Sending state 'Normal'
[11:45:23][D][sensor:093]: 'MHI-AC-Ctrl__2 Protection error state': Sending state 0.00000  with 0 decimals of accuracy
[11:45:23][W][component:214]: Component <unknown> took a long time for an operation (0.05 s).
[11:45:23][W][component:215]: Components should block for at most 20-30ms.
 `
PS
I apologize for the long logs. I don't know how to hide them)

ervee, 2023-11-15:
But does the vane controls and 3D auto still work?

ervee, 2023-11-15:
Yes. It was reverse engineered that these functions needs a bigger frame size to communicate with the AC unit and this is what you probably reverted with the commented line in mhi_ac_ctrl.h

AlexzeyK, 2023-11-15:
I agree with you. 
I guessed that these frame size changes were made to implement 3D Auto and left/right steering. 
But these changes broke the integration (

Dennis-Q, 2024-01-09:
@AlexzeyK Thank you for sharing this. I was a bit surprised that the amount of posts/reports with this issue are not bigger than they actually are. Now I probably found why.

It seems it has to do with the model/compatibility of the model itself. For newer models, I think the code is perfectly fine. They indeed support the bigger frame size of 33 (the line you commented out). The older models do not seem to stick with the 'latest standard' and cannot handle this bigger frame size.

I first had this project implemented for a very long time on my own airco-units (installed in 2016) with indoor units SRK35ZM-S and SRK50ZM-S (basically, I expect all SRK xx ZM-S to be the same). It has worked fine but just like in your case, with the latest commits it stopped working. The 'loop error code: -2' like messages were there like crazy which also somewhat indicates an issue with the communication of the units. Even though this happens to me every now and then (even with the old code), it was never like the flooding I get when the frame_size is set to 33.

Last week I installed the same module in a newer unit: SRK35ZS-W. I discovered that the exact same code, without changing/commenting the frame_size line, is working fine for this unit. So it is a compatibility issue with just specific units.

Now, although this is a theory; my unit never officially supported the 'WF-RAC' official wifi modules of MHI. Apart from the fact that it is very hard to reach the 'CNS' interface port, I imagine it is also because the units are not compliant with the 'latest' interface standard and therefore, the module doesn't work properly with this units. This probably counts for all the units without official 'WF-RAC' support; it might work, but depending on the model (and possibly it's internal firmware) the latest release will not work due to the unsupported frame_size.

@ginkage  is this something maybe to add to the README to make users aware?

I think it's good to get a list of units that face this issue. @AlexzeyK maybe you can share the indoor-unit you have that faces this issue as well?

Thank you all for contributing to this project!

AlexzeyK, 2024-01-09:
@Dennis-Q  Thank you for communicating about this issue. I mentioned my block model in the first post.
[image]
It's hard to accept that my model is older than 2016)
Let me know if you need my help (test the code, sniff the block dialog)

RobertJansen1, 2024-01-09:
https://github.com/RobertJansen1/MHI-AC-Ctrl-ESPHome/tree/legacy_support

i patched the files and added an option to configure this setting from the yaml file. if this can be tested, it would be greatly appreciated and i can add documentation for it if this works :)

Dennis-Q, 2024-01-09:
@AlexzeyK  thank you for sharing (again). I honestly have read your post a long time ago and forgot you did have the model in there. I understand your frustration regarding the 'old model'.

@RobertJansen1 
I added the 'globals' part you added to my own esphome yaml and used the .h and .cpp files from your legacy_support branch. It is working exactly as you would expect (identical to the manual change); with '33' value the log is flooded with the loop error: -2, but when set to '20', it is working perfectly fine (apart from some of the features).

So this means we won't have left-right and up/down, but it seems that the up/down is also incorrectly reported. Is this something that easily can be debugged? I understand it is somewhat unrelated.
Also; I wonder if we should add something more (maybe to the esphome yaml) to not expose 3D Auto and left-right options when they won't work anyway (e.g. frame size set to 20).

Thank you for working on this so quickly. I thought I suggested to implement such a feature in the code with my comment from this morning, but it seems I removed that part haha. Feels like coincidence that you started working on this yesterday already, looking at the commits of the branch :)

AlexzeyK, 2024-01-10:
@RobertJansen1  I just tried your version with legacy_support. I apologize for the delay - the subject is in the workspace)
The behavior of my SRK 63 ZSPR-S  completely matches the description [Dennis-Q](https://github.com/Dennis-Q)
also with value 33 i get -2 error.
likewise with a value of 20 I get working integration except for 3D control and horizontal blind control.

RobertJansen1, 2024-01-11:
oke, did some yaml magic to have one line comment / uncomment the new features, or enable legacy mode. it also removes 3d auto and vanes lr for unsupported models, and updated the readme.

Dennis-Q, 2024-01-11:
Great work, looking good and I think all is clear. I did do some suggestion in my comment of your pull request. Thanks for the effort taken!

Dennis-Q, 2024-01-15:
@heartkingz the pull request has not been committed yet. If you can't wait, you can use @RobertJansen1 his repo for now:
https://github.com/RobertJansen1/MHI-AC-Ctrl-ESPHome/tree/master

heartkingz, 2024-01-15:
> @heartkingz the pull request has not been committed yet. If you can't wait, you can use @RobertJansen1 his repo for now: https://github.com/RobertJansen1/MHI-AC-Ctrl-ESPHome/tree/master
> 
> And indeed; changing the framesize in the .h file from 33 to 20 (or commenting out the line) will probably fix your issue.

Ah i was just looking for the wrong repo lol. Thanks for clarifying!

RobertJansen1, 2024-01-18:
> I have one more question for you: where and what should I write in the code so that the air conditioners turn on with certain parameters? After the power supply is restored, the devices should turn on:
> 
> 1. cooling mode
> 2. The power of the flow is maximum
> 3. Shoulder blades (vertical/horizontal) - swaying.
> 
> This should direct the module MHI-AC-Ctrl if there is no connection to the server (i.e. in offline mode) or before connecting to the server.

i think you could do with some magic in the on_boot parameters of esphome, and maybe also add a disconnected. that is offtopic for this issue though, if you want more help with that, please create a new issue.

i will close this issue, as the problem was solved and merged into master.

closed 2024-01-18 by AlexzeyK