ginkage/MHI-AC-Ctrl-ESPHome #81: Total energy usage sensor does not update

heartkingz, 2024-02-21:
Does somebody know why the total energy used sensor does not update and stays 0.00 kWh? 
Do I need to change the update interval or something?

arpiecodes, 2024-03-30:
AFAIK the internal energy consumption counter updates only per x kwh of energy consumed, and also resets on power loss/counter wrap. If you use the AC a lot it should eventually start increasing, but very slow.

It is not a really reliable/precise way of measuring. If you're really looking for a power measurement, you can use volts * current reported as a better rough real-time estimate, or use CT clamps/other energy monitoring solution.

heartkingz, 2024-03-30:
Thanks for clarifying! Ended up making my own sensors with volts (DSMR5 source) * current. Works perfectly. 
Only downside is that i have a setup with one outside unit and two inside splits. When I turn on one AC unit, the other draws power as well because they are connected to the same outside unit right? There is no way to tackle this so I can individually monitor the usage?

arpiecodes, 2024-03-30:
> Thanks for clarifying! Ended up making my own sensors with volts (DSMR5 source) * current. Works perfectly. Only downside is that i have a setup with one outside unit and two inside splits. When I [...]

That's a good question. I'm guessing it is indeed because the energy consumption values are actually coming from the outside unit, not the inside unit (in terms of measurement), same as compressor frequency is the same for both.

You could maybe come up with some creative way, e.g. do some math on the composite value. If one unit on, then consumption = for that unit alone. If both units on, consumption divided by 2. It's still not precise, but at least gives some idea of where the consumption went to (the total is still somewhat precise).

heartkingz, 2024-03-30:
Good options indeed! That's pretty advanced and not super accurate I guess... I'll leave it for now and maybe just make two sensors for both my outdoor units. (I have three indoor units with two outdoor units). Thanks for sharing your ideas!

RobertJansen1, 2024-04-04:
> Thanks for clarifying! Ended up making my own sensors with volts (DSMR5 source) * current. Works perfectly. Only downside is that i have a setup with one outside unit and two inside splits. When I [...]

Can you make a pull request of this code, or share the code so we can include this?

heartkingz, 2024-04-04:
> Can you make a pull request of this code, or share the code so we can include this?

It all happens in Home Assistant itself. So no adjustments made on the ESPHome side. 

I made a template sensor for each inside unit: 

`template:
    sensor:
      - name: 'Stroomverbruik Airco Woonkamer'
        unit_of_measurement: W
        device_class: power
        state_class: measurement
        state: "{{ ( states('sensor.voltage_phase_l1') | float * states('sensor.airco_woonkamer_current_power_usage') | float ) | round(0) }}"
        availability: "{{ has_value('sensor.voltage_phase_l1') and has_value('sensor.airco_woonkamer_current_power_usage') }}"`

Once these are created, I created some helpers. First a couple of Riemann Sum Integral sensors and after that some Utility Meter sensors to collect data over a period of time. I have automated these sensors to reset to 0 after a week or a month for example. Works pretty well!

I'm no programmer or Github expert by any means. I cherry pick some bits and pieces and ask around for some best practices. Normally I just figure it out with some willpower and imagination 😃

EvoSems, 2024-04-26:
> AFAIK the internal energy consumption counter updates only per x kwh of energy consumed, and also resets on power loss/counter wrap. If you use the AC a lot it should eventually start increasing, [...]

Hi. is it possible for you to provide a template guide to do this? many thanks