ginkage/MHI-AC-Ctrl-ESPHome #82: [W][component:214]: Component took a long time for an operation (0.06 s). issue

EvoSems, 2024-02-22:
I have a new AC. and am getting a lot of errors in the logs. is this something to worry about?
my other new ac. doesn't show these a=errors at all. no errors at all

INFO Upload took 4.51 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from lounge_ac.local using esphome API
WARNING Can't connect to ESPHome API for lounge_ac.local: Error resolving IP address: [Errno 8] nodename nor servname provided, or not known (APIConnectionError)
INFO Trying to connect to lounge_ac.local in the background
INFO Successfully connected to lounge_ac @ 192.168.31.81 in 0.036s
INFO Successful handshake with lounge_ac @ 192.168.31.81 in 0.377s

[17:13:19][W][component:214]: Component took a long time for an operation (0.06 s).
[17:13:19][W][component:215]: Components should block for at most 20-30ms.
[17:13:19][W][component:214]: Component took a long time for an operation (0.06 s).
[17:13:19][W][component:215]: Components should block for at most 20-30ms.
[17:13:19][W][component:214]: Component took a long time for an operation (0.06 s).
[17:13:19][W][component:215]: Components should block for at most 20-30ms.
[17:13:19][W][component:214]: Component took a long time for an operation (0.06 s).
[17:13:19][W][component:215]: Components should block for at most 20-30ms.
[17:13:19][W][component:214]: Component took a long time for an operation (0.06 s).
[17:13:19][W][component:215]: Components should block for at most 20-30ms.
[17:13:19][W][component:214]: Component took a long time for an operation (0.06 s).
[17:13:19][W][component:215]: Components should block for at most 20-30ms.
[17:13:19][W][component:214]: Component took a long time for an operation (0.06 s).
[17:13:19][W][component:215]: Components should block for at most 20-30ms.
[17:13:20][W][component:214]: Component took a long time for an operation (0.06 s).
[17:13:20][W][component:215]: Components should block for at most 20-30ms.
[17:13:20][W][component:214]: Component took a long time for an operation (0.08 s).
[17:13:20][W][component:215]: Components should block for at most 20-30ms.
[17:13:20][W][component:214]: Component took a long time for an operation (0.06 s).
[17:13:20][W][component:215]: Components should block for at most 20-30ms.
[17:13:20][W][component:214]: Component took a long time for an operation (0.06 s).
[17:13:20][W][component:215]: Components should block for at most 20-30ms.
[17:13:20][W][component:214]: Component took a long time for an operation (0.06 s).
[17:13:20][W][component:215]: Components should block for at most 20-30ms.
[17:13:20][W][component:214]: Component took a long time for an operation (0.06 s).
[17:13:20][W][component:215]: Components should block for at most 20-30ms.
[17:13:20][W][component:214]: Component took a long time for an operation (0.06 s).
[17:13:20][W][component:215]: Components should block for at most 20-30ms.
[17:13:21][W][component:214]: Component took a long time for an operation (0.06 s).
[17:13:21][W][component:215]: Components should block for at most 20-30ms.
[17:13:21][W][component:214]: Component took a long time for an operation (0.06 s).
[17:13:21][W][component:215]: Components should block for at most 20-30ms.
[17:13:21][W][component:214]: Component took a long time for an operation (0.06 s).
[17:13:21][W][component:215]: Components should block for at most 20-30ms.

p.s thanks for all your hard work

EvoSems, 2024-02-23:
figured that one out.

  logs:
    component: ERROR

RobertJansen1, 2024-03-26:
This is not a loop error, but a long time for an operation, more info can be found at https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/issues/61

closed 2024-03-26 by RobertJansen1