ginkage/MHI-AC-Ctrl-ESPHome #85: mhi_ac_ctrl_core.loop error: -3 on newly flashed Wemos D1 Mini

northirid, 2024-03-21:
I have recently got all the parts together to make 6x MHI-AC-Ctrl PCBs (from v2.2 of the files on [here](https://github.com/absalom-muc/MHI-AC-Ctrl/)). I've assembled them all, flashed them all this morning, and went to try the first one, assuming that the error I was receiving was due to not being installed in the in-wall unit.

Home assistant can talk to and from the ESP, but there appears to be no communication between the ESP and the MHI unit when connected to the CNS port.

I am receiving mhi_ac_ctrl_core.loop error: -4 when the ESP is not plugged into the MHI-AC_Ctrl board, and mhi_ac_ctrl_core.loop error: -3 when I am connected to any of my 6 carrier boards.

I've read through the issues and tried changing the framesize, though that made for no change.  I'm struggling to work out the source of the -3 - can anybody assist?

RobertJansen1, 2024-03-21:
what units do you use? maybe as an alternative, try flashing the non-esphome version to see if the communication on that one is working (i don't expect it, and i do expect a hardware issue to be honest)

northirid, 2024-03-21:
I'm using these [USB-C D1 Mini clones](https://www.aliexpress.com/item/1005005977413920.html?spm=a2g0o.order_detail.order_detail_item.3.5d45f19ccYf4q1) - specifically the 8266, not the 32s.

They're into a DXK12ZSA-W wall unit, which appears to be an Australia specific model.  With that said, the PCB looks all but identical to those I've seen successfully using MHI-AC-CTRL.

[image]

>  try flashing the non-esphome version

I'm slightly embarrassed to admit I'm not sure how to.

northirid, 2024-03-21:
I was [testing using this](https://github.com/absalom-muc/MHI-AC-Ctrl/issues/84#issuecomment-1097201065), and found I wasn't getting 3.3V on the D1's 3.3V pin.

Reflowed all of the pin headers on the D1 board and it worked.  Hardware issue, and (me specifically) issue. :)

Thank you checking in though!

closed 2024-03-21 by northirid