ginkage/MHI-AC-Ctrl-ESPHome #88: setting temperature to .5 degrees doesn't work

RobertJansen1, 2024-04-04:
Currently, setting the temperature to 19.5 just sends a 19 degrees value to the airco unit as the units don't support half degree target temperatures.

in the upstream project, this is worked-around by mocking the temperature reading of the internal sensor and modifying the temp. we should do the same some time. the solution in #87 allows for this to be extended and should be investigated

locussss, 2024-11-30:
Hi Robert, this is still an open issue or is resolved?

RobertJansen1, 2025-07-07:
https://github.com/RobertJansen1/MHI-AC-Ctrl-ESPHome/tree/low_target
low target temp and 0.5 degree precision _should_ be possible from this tree, if someone has verified i will create a merge request