ginkage/MHI-AC-Ctrl-ESPHome #90: Can't set it up.

Apacalipto, 2024-05-19:
Noobie here.

Want to use the software, and its my first time working with an esp8266 device.

I'm getting the following error over and over:

[22:57:08][W][component:237]: Component <unknown> took a long time for an operation (101 ms).
[22:57:08][W][component:238]: Components should block for at most 30 ms.
[22:57:09][W][mhi_ac_ctrl:143]: mhi_ac_ctrl_core.loop error: -4
[22:57:09][W][component:237]: Component <unknown> took a long time for an operation (101 ms).
[22:57:09][W][component:238]: Components should block for at most 30 ms.
[22:57:09][W][mhi_ac_ctrl:143]: mhi_ac_ctrl_core.loop error: -4
[22:57:09][W][component:237]: Component <unknown> took a long time for an operation (101 ms).
[22:57:09][W][component:238]: Components should block for at most 30 ms.
[22:57:09][W][mhi_ac_ctrl:143]: mhi_ac_ctrl_core.loop error: -4
[22:57:09][W][component:237]: Component <unknown> took a long time for an operation (101 ms).
[22:57:09][W][component:238]: Components should block for at most 30 ms.

When i do esphome logs lr_mhi_ac_ctrl.yaml in the CMD.

i already changed the yaml to the legacy code, removed the .esphome folder and ran the following command again:

esphome lr_mhi_ac_ctrl.yaml run.

After that i pick 1 (usb device) to install.

Tweaks i made to the Yaml.
I added an password to the AP.
I changed the board type (this was the name my board got from home assistant when i flashed it before so i thought this is the right one, but i switched to flashing through cmd because i couldn't find out how to add the framesize files etc. to the homeassistant installation method. 

Wifi SSD and password in string.

My Yaml:

# Version 2.1
substitutions:
  # Unique device ID in HA
  deviceid: "mhi_ac_ctrl"
  # Unique device name in HA
  devicename: "MHI-AC-Ctrl"

esphome:
  name: ${deviceid}
  friendly_name: ${devicename}
  platform: ESP8266
  board: esp01_1m
  platformio_options:
    # Run CPU at 160Mhz to fix mhi_ac_ctrl_core.loop error: -2
    board_build.f_cpu: 160000000L
  includes:
    - mhi_ac_ctrl.h
    - MHI-AC-Ctrl-core.h
    - MHI-AC-Ctrl-core.cpp

packages:
  # Comment if you encounter mhi_ac_ctrl_core.loop error: -2 errors and uncomment the legacy_framesize file
  #mhi-ac-ctrl: !include conf/large_framesize.yaml
  # Uncomment this to allow for a working configuration on floor units or older units
  mhi-ac-ctrl: !include conf/legacy_framesize.yaml

wifi:
  ssid: "Deco-wifi-42"
  password: "Tracker!X"

  ap:
    ssid: "Wifi Fallback Hotspot"
    password: "50NSos29Ac0B"
logger:
  level: DEBUG
  baud_rate: 0
    
ota:

button:
  - platform: restart
    name: Restart
    entity_category: diagnostic

api:
  reboot_timeout: 0s
  services:
    # Call the set_api_room_temperature service from HA to override the room temperature
    # If a new value has not been received after room_temp_api_timeout seconds, it will fall back to internal sensor
    - service: set_api_room_temperature
      variables:
        value: float
      then:
        - lambda: |-
            return ((MhiAcCtrl*)id(${deviceid}))->set_room_temperature(value);
    # Call the set_vanes service from HA to set the vane position
    # Needed because the ESPHome Climate class does not support this natively
    # Possible values: 1-4: static positions, 5: swing, 0: unknown
    - service: set_vanes
      variables:
        value: int
      then:
        - lambda: |-
            return ((MhiAcCtrl*)id(${deviceid}))->set_vanes(value);

sensor:
  - platform: uptime
    name: Uptime
  - platform: wifi_signal
    name: WiFi Signal
    update_interval: 60s

text_sensor:
  - platform: version
    name: ESPHome Version
  - platform: wifi_info
    ip_address:
      name: IP
    ssid:
      name: SSID
    bssid:
      name: BSSID

Hope somebody here can help me troubleshoot this issue :). Don't know how to proceed and it's kind of frustrating to be stuck.

Thanks for developing this integration by the way :)

arpiecodes, 2024-05-19:
You need to connect it to the AC. The error indicates it has trouble communicating with the airconditioner. Connect it up and the error should go away.

Apacalipto, 2024-05-20:
connected it. But now i get this:

C:\Users\knobb\Downloads\MHI-AC-Ctrl-ESPHome-master\MHI-AC-Ctrl-ESPHome-master>esphome logs lr_mhi_ac_ctrl.yaml
INFO ESPHome 2024.5.0
INFO Reading configuration lr_mhi_ac_ctrl.yaml...
WARNING 'mhi_ac_ctrl': Using the '_' (underscore) character in the hostname is discouraged as it can cause problems with some DHCP and local name services. For more information, see https://esphome.io/guides/faq.html#why-shouldn-t-i-use-underscores-in-my-device-name
INFO Starting log output from mhi_ac_ctrl.local using esphome API
WARNING Can't connect to ESPHome API for mhi_ac_ctrl.local: Error resolving IP address: [Errno 11001] getaddrinfo failed (APIConnectionError)
INFO Trying to connect to mhi_ac_ctrl.local in the background

So it doesn't seem online? The wifi password and name are in "string" quotes. Is that an issue?

closed 2024-09-15 by RobertJansen1