ginkage/MHI-AC-Ctrl-ESPHome #95: [Feature Request] Self Cleaning - Fan Mode after Cooling / Drying

Ghostrider-1911, 2024-06-27:
Hello there, 
first i wanted to thank you for your great work! 

When i shut off the AC after cooling / drying with the original remote, the Fan keeps blowing 2h and Vanes are nearly shut.
This will dry the indoor unit to prevent bacteria etc. 

Is it possible to integrate this into this Project? I think many People could use this. 
[image]

heartkingz, 2024-07-30:
Good idea! Came across this feature and enabled it via the remote. It only works if you turn off the unit via the remote.

RobertJansen1, 2024-08-22:
I have read some forum messages of people adding this to Home Assistant (i did this in NodeRed). i am not sure if we want the unit to think for itself when home automation shuts it, but if you have an idea on how to implement it and document it clearly, we can consider it

Maxxxel, 2025-01-06:
i need a solution. I use the AC mostly for Dry and the room stinks like wet pants afterwards.

arpiecodes, 2025-01-06:
There is simply no solution available through the current MHI-Ac-Ctrl hardware to achieve this. Also, adding in logic on the module itself to kind of polyfill the feature in, makes no sense.

What you could do if you're a bit handy is add an IR transmitter module on one of the still available ESP-01's pins. Then you can emulate the remote key presses and turn off the AC that way (so it triggers the self-clean).

This will probably also allow for the other stuff being requested (like night setback/silent mode/turbo) over and over (but will never work due to SPI communication protocol limitations).

Something like this;

[image]

RobertJansen1, 2025-01-06:
an alternative would be to create an automation, if ac unit changed from heat / cool to off, set fan only for x minutes

Maxxxel, 2025-01-06:
I think an IR transmitter will be the way to go... But thinking about opening the case again...bro. they are so close under the ceiling 😅😅 it took 45 minutes last time.

RobertJansen1, 2025-01-06:
also i think the feature gets disabled as soon as something is connected through the SPI interface, i think remembering some official wifi unit users also wanting the fan feature, but were unable to (as it also doesn't work when the unit is controlled by the remote)

ajpalm, 2025-07-14:
Depending on the remote, to enable the self clean aka drying mode on the remote, you press a button with a paper clip to toggle the feature on/off. So its on all the time if enabled or at least on my remotes that is how you enabled it. You don't enabled this mode each time you use the unit.

So this could just be a toggle in ESPHome/HA to enable the clean mode when the unit turns off. The entire drying operation is handled by the unit when you turn it off.

It should aslo be noted this mode only runs when the unit has been in a cooling mode. So won't run when heating and turning the unit off.

I believe it is a different IR code for COOL/DRY/ON modes that is sent via the remote to tell the unit to do the drying cycle when turned off.

So should be able to be rolled in to keep everyone happy via a toggle.

arpiecodes, 2025-08-15:
Asked GPT 5 the same question; in my case it said there's no such evidence to believe this is part of the SPI protocol. Sorry to say, but it seems to have hallucinated support for this. It's a big issue with AI nowadays.

ajpalm, 2025-08-15:
Worth testing at least, easy to verify, turn to cool mode, then turn off. If the fan continues to run and the light flashes, it worked.

It is part of the IR remote code that is sent, so no reason the wall remotes can't do the same. Just need to work out which bit to flip.

All the button on the remote is doing is changing the IR code that is sent to flip the right bits.