ginkage/MHI-AC-Ctrl-ESPHome #97: Airconditioning not responding to commands

Hulsegge, 2024-08-22:
Hi, 

My air conditioning(SRK35SW-Z) is not responding, the device appears in homeassistant etc but all sensensors have the value state unknown.  I'm using the following [board()](https://www.nodo-shop.nl/nl/ons-assortiment/228-espac-wifi-mitsubishi-heavy-industries-interface.html). Pressing any button in homeassistant has zero effect on the air conditioning.

I guess it has something to do with the but i cannot discover what this means. 

**mhi_ac_ctrl_core.loop error: -3**

INFO ESPHome 2024.8.0
INFO Reading configuration /config/esphome/esphome-web-56a911.yaml...
WARNING 'mhi_ac_ctrl_1': Using the '_' (underscore) character in the hostname is discouraged as it can cause problems with some DHCP and local name services. For more information, see https://esphome.io/guides/faq.html#why-shouldn-t-i-use-underscores-in-my-device-name
INFO Starting log output from 192.168.1.109 using esphome API
INFO Successfully connected to mhi_ac_ctrl_1 @ 192.168.1.109 in 0.040s
INFO Successful handshake with mhi_ac_ctrl_1 @ 192.168.1.109 in 0.313s
[08:46:29][I][app:100]: ESPHome version 2024.8.0 compiled on Aug 21 2024, 18:35:25
[08:46:29][C][wifi:600]: WiFi:
[08:46:29][C][wifi:428]:   Local MAC: C8:C9:A3:56:A9:11
[08:46:29][C][wifi:433]:   SSID: [redacted]
[08:46:29][C][wifi:436]:   IP Address: 192.168.1.109
[08:46:29][C][wifi:439]:   BSSID: [redacted]
[08:46:29][C][wifi:441]:   Hostname: 'mhi_ac_ctrl_1'
[08:46:29][C][wifi:443]:   Signal strength: -60 dB ▂▄▆█
[08:46:29][C][wifi:447]:   Channel: 6
[08:46:29][C][wifi:448]:   Subnet: 255.255.255.0
[08:46:29][C][wifi:449]:   Gateway: 192.168.1.100
[08:46:29][C][wifi:450]:   DNS1: 192.168.1.100
[08:46:29][C][wifi:451]:   DNS2: 0.0.0.0
[08:46:29][W][mhi_ac_ctrl:143]: mhi_ac_ctrl_core.loop error: -3
[08:46:29][W][component:237]: Component <unknown> took a long time for an operation (101 ms).
[08:46:29][W][component:238]: Components should block for at most 30 ms.
[08:46:29][C][logger:185]: Logger:
[08:46:29][C][logger:186]:   Level: DEBUG
[08:46:29][C][logger:188]:   Log Baud Rate: 0
[08:46:29][C][logger:189]:   Hardware UART: UART0
[08:46:29][W][mhi_ac_ctrl:143]: mhi_ac_ctrl_core.loop error: -3
[08:46:29][W][component:237]: Component <unknown> took a long time for an operation (101 ms).
[08:46:29][W][component:238]: Components should block for at most 30 ms.
[08:46:29][W][mhi_ac_ctrl:143]: mhi_ac_ctrl_core.loop error: -3
[08:46:29][W][component:237]: Component <unknown> took a long time for an operation (101 ms).
[08:46:29][W][component:238]: Components should block for at most 30 ms.
[08:46:29][C][uptime.sensor:033]: Uptime Sensor 'Uptime'
[08:46:29][C][uptime.sensor:033]:   Device Class: 'duration'
[08:46:29][C][uptime.sensor:033]:   State Class: 'total_increasing'
[08:46:29][C][uptime.sensor:033]:   Unit of Measurement: 's'
[08:46:29][C][uptime.sensor:033]:   Accuracy Decimals: 0
[08:46:29][C][uptime.sensor:033]:   Icon: 'mdi:timer-outline'
[08:46:29][C][uptime.sensor:034]:   Type: Seconds
[08:46:29][W][mhi_ac_ctrl:143]: mhi_ac_ctrl_core.loop error: -3
[08:46:29][W][component:237]: Component <unknown> took a long time for an operation (101 ms).
[08:46:29][W][component:238]: Components should block for at most 30 ms.
[08:46:29][C][template.select:065]: Template Select 'Fan Control Left Right'
[08:46:29][C][template.select:065]:   Icon: 'mdi:arrow-left-right'
[08:46:29][C][template.select:066]:   Update Interval: 60.0s
[08:46:29][C][template.select:069]:   Optimistic: YES
[08:46:29][C][template.select:070]:   Initial Option: Left
[08:46:29][C][template.select:071]:   Restore Value: NO
[08:46:29][W][mhi_ac_ctrl:143]: mhi_ac_ctrl_core.loop error: -3
[08:46:29][W][component:237]: Component <unknown> took a long time for an operation (101 ms).
[08:46:29][W][component:238]: Components should block for at most 30 ms.
[08:46:30][C][template.select:065]: Template Select 'Fan Control Up Down'
[08:46:30][C][template.select:065]:   Icon: 'mdi:arrow-up-down'
[08:46:30][C][template.select:066]:   Update Interval: 60.0s
[08:46:30][C][template.select:069]:   Optimistic: YES
[08:46:30][C][template.select:070]:   Initial Option: Up
[08:46:30][C][template.select:071]:   Restore Value: NO
[08:46:30][W][mhi_ac_ctrl:143]: mhi_ac_ctrl_core.loop error: -3
[08:46:30][W][component:237]: Component <unknown> took a long time for an operation (101 ms).
[08:46:30][W][component:238]: Components should block for at most 30 ms.
[08:46:30][C][template.switch:068]: Template Switch '3D Auto'
[08:46:30][C][template.switch:070]:   Icon: 'mdi:video-3d'
[08:46:30][C][template.switch:090]:   Restore Mode: always OFF
[08:46:30][C][template.switch:057]:   Optimistic: YES
[08:46:30][W][mhi_ac_ctrl:143]: mhi_ac_ctrl_core.loop error: -3
[08:46:30][W][component:237]: Component <unknown> took a long time for an operation (101 ms).
[08:46:30][W][component:238]: Components should block for at most 30 ms.
[08:46:30][C][mhi_ac_ctrl:148]: MHI-AC-Ctrl Climate 'MHI-AC-Ctrl-woning-boven'
[08:46:30][C][mhi_ac_ctrl:149]:   Min. Temperature: 18.0°C
[08:46:30][C][mhi_ac_ctrl:150]:   Max. Temperature: 30.0°C
[08:46:30][C][mhi_ac_ctrl:151]:   Supports HEAT: YES
[08:46:30][C][mhi_ac_ctrl:152]:   Supports COOL: YES
[08:46:30][W][mhi_ac_ctrl:143]: mhi_ac_ctrl_core.loop error: -3
[08:46:30][W][component:237]: Component <unknown> took a long time for an operation (101 ms).
[08:46:30][W][component:238]: Components should block for at most 30 ms.

RobertJansen1, 2024-08-22:
loop error -3 is usually an issue with the board or pinout (you can change the pins in the yaml) 
another thing you can try is to change to the legacy framesize, which works for older models of the airco unit since not all units support the larger framesize. the error i would expect for this issue though is loop error -2, so if the legacy framesize also fixes the -3 error, i would like to know so we can update the yaml / documentation :)

Hulsegge, 2024-08-22:
Hi Robert, thanks for your fast response.
legacy framesize i already tried. 
can you show me how to change the pins in yaml?

RobertJansen1, 2024-08-22:
the pins are not in the yaml, but can be found in the

Hulsegge, 2024-08-22:
#define SCK_PIN  14
#define MOSI_PIN 13
#define MISO_PIN 12

so swapping pin 12 and 14?

RobertJansen1, 2024-08-22:
well, that depends on the board you are using, and the number of the pins that connect to the boad in the AC. just to be sure, is the board with esp in the AC already? and what esp are you using?

Hulsegge, 2024-08-22:
Yes its inside, well i still have all open but its connected to the AC. i use the Wemos D1 Mini ESP8266 combined with 
[](https://www.nodo-shop.nl/nl/ons-assortiment/228-espac-wifi-mitsubishi-heavy-industries-interface.html)

RobertJansen1, 2024-08-30:
did you manage to get it to work?

Dennis-Q, 2024-10-04:
> Yes its inside, well i still have all open but its connected to the AC. i use the Wemos D1 Mini ESP8266 combined with [board](https://www.nodo-shop.nl/nl/ons-assortiment/228-espac-wifi-mitsubishi-heavy-industries-interface.html)
I have been using this board in 3 units now and the pinout of this project is fine by default.

Please make sure you did the soldering correctly and all connections are OK.

Drscrote, 2024-10-09:
HI mines doing a similar thing, all the sensors show as "Unknown" and it doesn't respond.

My log is basically silent, this is 5 minutes worth. Oddly I can see the state of the AC in the ESPhome integration in HA, including mode and room temperature, but none of the sensors. 

I'm using a [WEMOS D1 Mini V4.0.0 ESP8266 4MB](https://www.aliexpress.com/item/1005006246661815.html?spm=a2g0o.order_list.order_list_main.53.103e1802dXZYpN)
The pinout seems to match the one in the core file.

I've tried reinstalling the code to the board, clearing out the files from the ESPhome folder in HA and reloading them. 

Hoping not to have to dismantle the unit again to check my soldering!

Tom

INFO Starting log output from 192.168.68.116 using esphome API
INFO Successfully connected to mitsubishi-kitchen-ac @ 192.168.68.116 in 19.379s
INFO Successful handshake with mitsubishi-kitchen-ac @ 192.168.68.116 in 0.218s
[16:36:53][I][app:100]: ESPHome version 2024.9.2 compiled on Oct  9 2024, 16:36:07
[16:37:39][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[16:37:56][W][mhi_ac_ctrl:161]: mhi_ac_ctrl_core.loop error: -2
[16:38:23][W][mhi_ac_ctrl:161]: mhi_ac_ctrl_core.loop error: -1
[16:38:33][W][mhi_ac_ctrl:161]: mhi_ac_ctrl_core.loop error: -2
[16:39:14][W][mhi_ac_ctrl:161]: mhi_ac_ctrl_core.loop error: -2
[16:40:22][W][mhi_ac_ctrl:161]: mhi_ac_ctrl_core.loop error: -1
[16:41:22][W][mhi_ac_ctrl:161]: mhi_ac_ctrl_core.loop error: -2
[16:42:47][W][mhi_ac_ctrl:161]: mhi_ac_ctrl_core.loop error: -2
[16:42:48][W][mhi_ac_ctrl:161]: mhi_ac_ctrl_core.loop error: -2
[16:43:56][W][mhi_ac_ctrl:161]: mhi_ac_ctrl_core.loop error: -2

Drscrote, 2024-10-09:
I suspect it might be the V4 board - I'll grab an older one.

ogizhelev, 2024-10-27:
Hi,

Did someone found solution on this one ? 

I have indoor unit SRF35ZS-W that I can control however the esp log is full of 
error : 

```
INFO ESPHome 2024.10.2
INFO Reading configuration /config/esphome/dining_room_ac.yaml...
INFO Starting log output from 10.10.4.106 using esphome API
INFO Successfully connected to dining-room-ac @ 10.10.4.106 in 0.038s
INFO Successful handshake with dining-room-ac @ 10.10.4.106 in 0.169s
[15:11:22][I][app:100]: ESPHome version 2024.10.2 compiled on Oct 27 2024, 14:54:34
[15:11:22][W][component:237]: Component MhiAcCtrl took a long time for an operation (61 ms).
[15:11:22][W][component:238]: Components should block for at most 30 ms.
[15:11:24][W][component:237]: Component MhiAcCtrl took a long time for an operation (56 ms).
[15:11:24][W][component:238]: Components should block for at most 30 ms.
[15:11:24][W][component:237]: Component MhiAcCtrl took a long time for an operation (61 ms).
[15:11:24][W][component:238]: Components should block for at most 30 ms.
[15:11:24][W][component:237]: Component MhiAcCtrl took a long time for an operation (63 ms).
[15:11:24][W][component:238]: Components should block for at most 30 ms.
[15:11:24][W][component:237]: Component MhiAcCtrl took a long time for an operation (64 ms).
[15:11:24][W][component:238]: Components should block for at most 30 
```

I did change the frame_size to 20 but the error is still there 
[image]

another unit - SRK25ZS-W works with no issues at all.

stefinuxha, 2024-11-01:
I don't see any error in yours. Your message about component took long i have it as well, but I don't believe is an issue.

I also installed today also mine (i have and old unit) and initially did not worked, but then with frame size 20 I was able to make it work.
[image]

FYI i destroyed 2x Wemos D1 mini, anyone has an idea why? could the wrong frame be the cause? I was not able to get 3,3V I read somewhere they use low quality 3,3 V converter which fails...my problem that it did 2 times after just few seconds connected to HVAC...