ginkage/MHI-AC-Ctrl-ESPHome #100: Version 3.0 not compiling

Kurisutian, 2024-09-22:
I just checked to switch to the component change made in v3.0 but for whatever reason the I cannot compile. I first had the `time` definition in my yaml which ultimately fail but also when being as close as possible to the example provided the compilation fails. I attached the log for the error. Couldn't figure out what went wrong here. 

[logs_klima-eg_run.txt](https://github.com/user-attachments/files/17089669/logs_klima-eg_run.txt)

Thanks and please let me know if anything else is needed to be provided from my end. Gladly willing to help here.

RobertJansen1, 2024-09-22:
Try cleaning up your build files

RobertJansen1, 2024-09-22:
I have seen some duplicate definition errors when writing the code, and all of the cases it was duplicate / remaining files, so maybe double check the esphome folder for lingering old versions of the cpp and h files, which should no longer be placed in the folder manually. You can also try building it as a new entity in esphome to prevent old files lingering around in the build directories

Kurisutian, 2024-09-22:
Thanks for all your tips and willingness to help. I just checked that the esphome folder does not contain any remaining file whatsoever. Unfortunately the error remains. Checked again after that and cleaned everything again. Am I missing another folder which might contain files other than the esphome one?

Kurisutian, 2024-09-23:
I checked today on another device after upgrading to the new Esphome version and it seems that it's working now. Nothing changed though, but it went through. So I'm closing this one. Thanks a lot for your help. Don't know what fixed it but it is working. 🤷‍♂️

jeroenleenarts, 2024-09-24:
Getting the exact same error here.

I run a Home Assistant Green with the ESP Addon, how can I check on a file level that the cached versions of these files are gone? A "clean" in ESPHome does not resolve this issue.

jeroenleenarts, 2024-09-24:
Did a local commandline build. And that compiles fine, so that is in line with @RobertJansen1's statement on left over files being on my ESPHome instance. Question is, do these files expire by themselves if clean does not work?

ervee, 2024-10-06:
I had this exact same issue and I could solve it like this:

- Do the "Clean Build Files" on the device in Home Assistant ESPHome.
- Use SSH or console or some way to login onto the device running Home Assistant.
  - Go to ESPHome data folder for the device: ```cd /mnt/data/supervisor/addons/data/*_esphome/build/_devicename_/```
  - Check folder content: ```ls -l```
  - Make sure you are in the correct folder for your device and there is a ```scr``` folder! Rename the folder: ```mv src src.old```
- Now try to install / build / update the device in Home Assistant ESPHome again.

jeroenleenarts, 2024-10-07:
https://github.com/esphome/issues/issues/6312

hmmbob, 2024-10-15:
Just ran into the same problem, cleaning all files manually through SSH resolved the issue.

💡 Should we pin this issue to the top of the issue list? I guess more people will run into this issue....

RobertJansen1, 2024-10-15:
I added this issue to the changelog (the comment from @ervee as that one seems to have the most complete steps to resolve the issue. sorry for the inconvenience everyone!

closed 2024-10-15 by RobertJansen1

1Dimension1, 2024-11-19:
When using Homeassistant OS you need to SSH into your installation:
- Install ssh addon (Advanced SSH & Web Terminal)
- Set password in configuration
- Connect to SSH
- `docker exec -it addon_5c53de3b_esphome sh`
- `cd /data/build/`

Then follow above comment (https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/issues/100#issuecomment-2395388853)