ginkage/MHI-AC-Ctrl-ESPHome #103: room_temp_api_timeout not working

th3sniper86, 2024-11-06:
I’m using an external temperature sensor to set the current room temperature using the service ‘set_api_room_temperature’. This works, but after one minute, the temperature is set back to the internal temperature sensor, despite the timeout has not passed.
[image]
Tested with AC off and in heating mode.

serrj-sv, 2024-11-14:
```
MhiAcCtrl:
  id: ${mhi_device_id}
  frame_size: ${frame_size}
  room_temp_timeout: 1800
```

th3sniper86, 2024-11-14:
> add room_temp_timeout in MhiAcCtrl section in your yaml:
> 
> ```
> MhiAcCtrl:
>   id: ${mhi_device_id}
>   frame_size: ${frame_size}
>   room_temp_timeout: 1800
> ```

Thanks!
This resolved the issue for me.

closed 2024-11-14 by th3sniper86

RobertJansen1, 2024-12-11:
Added to the example.yaml: https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/commit/309cd34828d0e55a7cfbcb62c637b5569f2ff8a3