ginkage/MHI-AC-Ctrl-ESPHome #105: Update process

northirid, 2024-11-14:
Hi all,

I currently have 7 units running on v2.0 of this.

Is there a straightforward way of updating to 3.0, or is the "easiest" way going to be to go through and match the values from oldunit.yaml into newunit.yaml and then overwrite oldunit.yaml with the updated code?

Theetjuh, 2024-12-17:
That’s what I did, it took some time, but all operational here.

Maxxxel, 2025-01-03:
Is there a detailed instruction somewhere?

I installed my unit like 8 mo ago. I have no idea what I need to do to update to 3.0

Please 🥺

RobertJansen1, 2025-01-06:
the easiest way i know, unfortunattely, is through the method described by @northirid 

other than the readme i have not done much on documenting this (there is some in the linked issue should you encounter issue) but it boils down to comparing the current esphome configuration in your home assistance with the new one, keep all unit-specific stuff (like names, passwords, ota and such) and replace everything else with the new code from the example yaml. if someone wants to write more detailed documentation, please feel free to create a pull request and i will happily review and merge it :)

RobertJansen1, 2025-04-04:
https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/tree/master?tab=readme-ov-file#everything-has-changed-how-do-i-update would this suffice for update documentation?

closed 2025-04-04 by RobertJansen1

northirid, 2025-04-05:
> would this suffice for update documentation?

Looks good to me!