ginkage/MHI-AC-Ctrl-ESPHome #108: Assistance - Update from MHI-AC-Ctrl to MHI-AC-Ctrl-ESPHome

beef-techau, 2024-11-23:
I've had a look at the open and closed issues. 
I have the OG MHI-AC-Ctrl running on 2 units, however I no longer have the computer or the configs I used to set them up (approx 4yrs ago) and have been a little out of touch with the smart home side of things as everything has been running fine. 
I primarily use HA so i like the idea of using ESPHome as everything is visible, where as at the moment its all just running through MQTT which i don't fully understand, i bumbled my way through it and its working Ha!

Is the hardware set up the same between these two projects? Can I simply set up a new D1 with this code, remove and replace the old one from the AC unit and off we go?

Apologies in advance if this is a dumb question to everyone else.

beef-techau, 2024-11-23:
yes it does
also i included a couple extra lines to use the DS1820, unsure if i need both but its working so i wont change it 
```
one_wire:
  - platform: gpio
    pin: GPIO4    # D2 on a Wemos D1 mini (8266)
sensor:
  - platform: dallas_temp
    name: temperature
    update_interval: 60s
```

RobertJansen1, 2025-01-06:
you could also connect a micro-usb cable to the current d1 and flash esphome on it, i think it would also be possible to update them through OTA, but that requires OTA to be configured on the initial deployment and might be more complex. if you need more assistance, please feel free to reopen the issue at any time :)

closed 2025-01-06 by RobertJansen1