ginkage/MHI-AC-Ctrl-ESPHome #109: Help requested: Fan Modes

jeroen-github, 2024-11-24:
Is there a way to set the fan modes from an automation in Home Assistant.
It's a little annoying that you not can change the fan modes from the main Climate interface, but you have to click the 3 dots top right (more information) to get to the Fan Modes.
Any help would be appreciated

northirid, 2024-11-25:
You can use Climate: Set Fan Mode

[image]

By looking at Dev Tools -> States -> your AC entity, you can see the modes available per here:

[image]

jeroen-github, 2024-11-25:
Thanks, this helped me a lot.

closed 2025-01-06 by RobertJansen1