ginkage/MHI-AC-Ctrl-ESPHome #110: Current power steering? Possible

locussss, 2024-11-30:
Hi!

I am really curious but not technically skilled enough but is it technically possible to control the temperature using "current power"? It is a sensor available in Home Assistant. 

I mean the amperage is not continuous but in discrete steps of about 0,28 A. Couldn't the target temperature be controlled with this to get an even better and smooth regulation?
Probably the answer is "No" because if it could, it would already been implemented, I guess. 

I have an example of the history of the amperage below.

[image]

I am just interested in what is possible with the module and maybe this is a silly question, tell me.

arpiecodes, 2024-11-30:
The current displayed is the current used by the compressor on the outside unit afaik. It's not possible to control the unit based on current, and I also have no idea why you would want to do that. The module modulates the compressor speed (and thus power consumption) based on how close the temperature is to your setpoint temperature.

Obviously you can raise the setpoint to a very high number, or lower it, to make it use more/less power, if that's what you're after.

locussss, 2024-11-30:
I thought it would be a silly question and it shows....

No, ultimately, I would like to be able to regulate the temperature with my external sensor and mhi-ac-ctrl even better then I can now, which is a whole level better then without this.

And from what I have read, it is possible with Daikin airco's to control the temperature also with power settings but I might be wrong.

RobertJansen1, 2025-07-16:
there is no way to directly control the MHI's compressor power / frequency based on a value like: use x Amps or run at x mhz. 

what you could do (with automations or more sophisticated logic) is frame the unit into running at the speed you want. my best bet would be to just send a target temp that is just slightly over or under the temp you want to achieve, and go with that. it is not really the logic i would want the esp to do though, so please share your automations once you have them :)

closed 2025-07-16 by RobertJansen1