ginkage/MHI-AC-Ctrl-ESPHome #112: ESP32S3 - ATOM S3 lite

fonske, 2024-12-03:
Hi All,

Not an issue, but to let know, i have made a new pcb with a M5stack Atom S3 lite instead of the Wemos D1 mini.
It is a bit faster and dual core, so bluetooth and wifi should be possible at the same time.
Also the atom has the possibility to have an extra sensor connected from M5stack, like the ENV IV, CO2, ENV pro, KMeter-iso thermocouple.
I used a few sensors and programmed them in the forked repo, in the atom branch.
https://github.com/fonske/MHI-AC-Ctrl-ESPHome/tree/atom
In the map photos there are some photos of the pcb.

If the sensor contains external temperature, it can be used to control the MHI. Its programmed in the sensor yaml.

@ginkage Can i merge this branch with your github repo? Thank you.

[image]

EvoSems, 2025-02-04:
Hi. very cool. keen to build! I have sent you an email. thanks

HpNoTiQ56, 2025-07-14:
Hi,
You deleted branch recently but I tried to use it with another esp32 but got a lot of loop error -2 even changing time for loop in program. Why did you deleted it? Too many errors or no use?

HpNoTiQ56, 2025-07-14:
Thank you!. How did you manage to get ride of the anoying loop on ESP32 (A/C are working great using esp8266). I'm using D1 MINI ESP3 using legacy board + redifining pins but log is flooded with loop error even when trying to increase loop timeout. I don't want webserver only bluetooth proxy!
I don't care about loop errors but sometimes the A/C doesn't execute the command sent over HA.