ginkage/MHI-AC-Ctrl-ESPHome #113: Using the Web Server Standalone Without Home Assistant

tech2mar, 2024-12-22:
Hi,

First of all, thank you for this great project!

I would like to use the MHI-AC-Ctrl module in a very specific way:

I want to control the air conditioning units exclusively via the web interface provided by the ESP module.
I do not intend to use Home Assistant, MQTT, or any other integrations—just the web server functionality to interact with the AC through a browser.
Before proceeding, I wanted to ask:

Is it feasible to use the web server as the sole interface for controlling the AC?
Are there any limitations or specific configurations required to make this work?
Your advice would be highly appreciated. Thank you for your hard work and for sharing this project!

Best regards,

RobertJansen1, 2024-12-22:
I have not tested it myself but I heard it is possible to run the unit stand alone, you do need to make sure it doesn't reboot when the api is ot connected, that's an esp home setting in the yaml file.

fonske, 2024-12-23:
The fan speed cannot be controlled in the webserver as it is part of the climate sensor.
I am working on adding code to have the fanspeed, quiet, low, medium, high and auto in a seperate select box so it is available to the webserver.

tech2mar, 2024-12-24:
That sounds good, thank you all for your help and support.

fonske, 2024-12-27:
@tech2mar 
I created a pull request and tested it.
https://github.com/ginkage/MHI-AC-Ctrl-ESPHome/pull/115

tech2mar, 2024-12-27:
This is very good, thank you for your time and support.