ginkage/MHI-AC-Ctrl-ESPHome #117: Reading failed: BAD_INDICATOR errno=11

backyardaussie, 2025-01-18:
just changed from 2.1 to 3.0

got it to install... but then it just logs out
```
INFO Successfully connected to loungeroom_ac @ 10.0.40.123 in 11.399s
INFO Successful handshake with loungeroom_ac @ 10.0.40.123 in 0.201s
[15:27:57][I][app:100]: ESPHome version 2024.12.4 compiled on Jan 18 2025, 15:27:35
[15:28:03][W][api.connection:107]: : Reading failed: BAD_INDICATOR errno=11
[15:28:05][W][api.connection:107]: : Reading failed: BAD_INDICATOR errno=11
[15:28:05][W][api.connection:107]: : Reading failed: BAD_INDICATOR errno=11
```

using DEBUG it is getting states but the device does not load in dashboard
```
[15:41:16][C][mhi_ac_ctrl:173]:   Supports HEAT: YES
[15:41:16][C][mhi_ac_ctrl:174]:   Supports COOL: YES
[15:41:16][C][restart.button:017]: Restart Button 'Restart'
[15:41:16][C][restart.button:017]:   Icon: 'mdi:restart'
[15:41:16][D][mhi_ac_ctrl:180]: received status=139 value=38 power=1
[15:41:16][D][sensor:093]: 'Indoor unit total run time': Sending state 3800.00000 h with 0 decimals of accuracy
[15:41:16][D][api:103]: Accepted 10.0.20.7
[15:41:16][W][api.connection:107]: : Reading failed: BAD_INDICATOR errno=11
[15:41:16][C][version.text_sensor:021]: Version Text Sensor 'ESPHome Version'
[15:41:16][C][version.text_sensor:021]:   Icon: 'mdi:new-box'
[15:41:16][C][captive_portal:089]: Captive Portal:

```
used to work great.. what changed or improved in v3 ??

RobertJansen1, 2025-01-20:
https://community.home-assistant.io/t/solved-reading-failed-bad-indicator-errno-11-after-esphome-2023-3-0-upgrade/550228

is your encryption key correct?

backyardaussie, 2025-01-20:
Not sure what key you mean?

But the device is working just fine, however it keeps logging this one item and I'm just curious as to what might be causing it.
There is not much in the logs to work with to debug why